xblax/flashforge_ad5m_klipper_mod #50: Filament runout sensor not working

johncotten, 2024-03-12:
The filament runout sensor does not appear to be on PB14. I also found reference of a PE1 but that does not appear to be correct either. 

Config in printer.base.cfg

[filament_switch_sensor runout_sensor]
pause_on_runout: False
switch_pin: !PB14
event_delay: 1.0

KaruroChori, 2024-03-12:
There were two filament runout sensors in the very first version, both disabled by default.
In theory, the one not used was disabled.

Could it be that different board revisions have one or the other connected?

consp, 2024-03-12:
Stock also seems to ignore the sensor. Dees anyone know it actually works?

jtenniswood, 2024-03-12:
Yes it does work on stock, I have experienced the printer parking when the sensor was triggered.

consp, 2024-03-12:
I guess there is some additional magic somewhere, diving into stock now to see what goes on.

consp, 2024-03-12:
Reloading the original cfg works, I can now detect it.

consp, 2024-03-12:
The fix is pushed to master, strange but true. I guess it's a configuration thing as the PE1 pin is not exposed on the 48 or 64 pin package.

closed 2024-03-12 by consp

xblax, 2024-03-12:
@consp Can you rename it to "_e1_sensor"? Then it should not show up in the Gui and confuse everyone.

consp, 2024-03-12:
> @consp Can you rename it to "_e1_sensor"? Then it should not show up in the Gui and confuse everyone.

sure, I'll amend it and force push for niceness

DougJoseph, 2024-03-24:
Thank you all for catching and fixing this! I just have one question. Where it says “pause at runout” is set to false, I would think that you would want it to pause at runout. Am I misunderstanding how that code is intended to function?

xblax, 2024-03-24:
> Thank you all for catching and fixing this! I just have one question. Where it says “pause at runout” is set to false, I would think that you would want it to pause at runout. Am I misunderstanding how that code is intended to function?

Yes, good point. I will try to test it.

xblax, 2024-03-24:
@DougJoseph 
I did now properly integrate the runout sensor with some macros:

[image]

This will be in the next release.

DougJoseph, 2024-03-25:
Wow you are amazing!! Thank you!!