xblax/flashforge_ad5m_klipper_mod #51: Add Klipper-Backup to image

jtenniswood, 2024-03-12:
Just wondering if it's possible to include Klipper Backup, as I've found it's really valuable in case of issue to have your config backed up to git.
https://staubgeborener.github.io/klipper-backup/

I've tried installing it but the script won't work, I'm sure it's something simple and unique about this environment that I can't figure out. Thanks!

KaruroChori, 2024-03-12:
To fit within the limitations of the SOC, we cannot run a full debian-based system, so you will find most of the arbitrary software you might want to run will not, even if compiled for this platform. Unless you are statically compiling a self-contained piece of software, using standard posix features at best, incompatibility is the likekly outcome.

The distribution provided in this mod is extremely limited compared to a more typical desktop linux environment.
In order to support that tool there are probably some services and dependecies missing.

xblax, 2024-03-12:
I think that these kind of features are probably to use-case specific to integrate them directly into the mod, because they could never be active as default. As was written above, the mod does not have Bash, Apt, Systemd etc due to being an embedded Linux. 

But in theory, the mod already includes all the necessary tools to implement such a feature (especially Git) and the G-Code shell extension could be useful as well. It should not be too difficult to build something around that and ideally maintain that in separate Github repo. We can maybe create a list of separately maintained mod extensions, that can be installed if needed.

jtenniswood, 2024-03-17:
I ended up working with ChatGPT to write this script...

`#!/bin/sh

# GitHub credentials and repository information
GITHUB_USERNAME="USER"
REPOSITORY_NAME="REPOSITORY"
GITHUB_EMAIL="EMAIL"
GITHUB_NAME="NAME"
GITHUB_TOKEN="TOKEN"

# Directory to upload
UPLOAD_DIR="/root/printer_data/config/"

# Directory to store Git repository files
GIT_DIR="/root/klipper-backup/upload/"

# Set the remote repository URL with the personal access token
REMOTE_REPO_URL="https://$GITHUB_USERNAME:$GITHUB_TOKEN@github.com/$GITHUB_USERNAME/$REPOSITORY_NAME.git"

# Navigate to the directory
cd "$UPLOAD_DIR" || { echo "Directory not found"; exit 1; }

# Initialize a new git repository with custom Git directory location
git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" init > /dev/null

# Set author identity
git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" config user.email "$GITHUB_EMAIL"
git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" config user.name "$GITHUB_NAME"

# Exclude specific file patterns
git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" ls-files | grep -Ev "\.bkp$" | xargs git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" add -u --

# Commit changes
git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" commit -m "Update files" > /dev/null

# Add the remote repository using GitHub token
git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" remote add origin "$REMOTE_REPO_URL" > /dev/null
git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" remote set-url origin "$REMOTE_REPO_URL" > /dev/null

# Force push changes to the remote repository
git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" push -f origin master

# Exit
exit 0`

It stores the git files in /root/klipper-backup/upload.

I used the existing tools the original script creates for making the ability to start a script and setup a macro.
I'm VERY far from being a developer, so this is just to help anyone else that might want to do something similar in the future (or if the team want to integrate something similar into a release).

jtenniswood, 2024-03-17:
> <rant>
>
> Script will probably work for you, but I have a distinct hate for ChatGPT
> created bloat since it is never done properly and regurgitates bad
> behaviour.
>
> git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" init > /dev/null
>
> Will error after the first time, doing output redirection to devnull is
> not desirable fault tollerance behaviour since everything else depends on
> this working. Should exit 1.
>
> git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" remote add origin "$REMOTE_REPO_URL" > /dev/null
>
> Will not detect invalid repo's and error every time after it has been done
> once.
>
> git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" remote set-url origin "$REMOTE_REPO_URL" > /dev/null
>
> Why on earth did it add, and then set again. It's simply ignoring errors
> and hoping for the best.
>
> git --git-dir="$GIT_DIR" --work-tree="$UPLOAD_DIR" push -f origin master
>
> Why on earth should it force push.
>
> This script should have a setup phase and an upload phase. The first one
> should be triggered only once.
>
> </rant>

Lol, spot on! Let's hope it channels enough anger for someone to fix it. I
found it funny that it just apologizes for every problem, it's like dealing
with a useless waiter.

xblax, 2025-09-16:
With the built-in features of the mod system (SSH, Git, Web based config via Mainsail/Fluidd) it's already very easy to to do backups as needed individually. 

Klipper config needs to be override on updates to stay compatible with changed components. Additional backup methods are not planned.

closed 2025-09-16 by xblax