xblax/flashforge_ad5m_klipper_mod #52: printer: UNLOAD_FILAMENT macro not configured

consp, 2024-03-12:
Klipperscreen wants the UNLOAD and LOAD_FILAMENT macros to be available. LOAD works but UNLOAD is missing.

KaruroChori, 2024-03-12:
The main issue is how we want to unload the filament?
The "flashforge approach" is to cut the filament, while most printers would just perform some fancy retraction. 
My understanding is that the stock firmware never retracts molten material back through the hot-end.

They designed all the original macros and procedures around the idea "filament must go only one way" and it seems like that was quite deliberate.

consp, 2024-03-12:
Interesting, could just put in an empty macro. I'll have a look if it can show a message or something telling the user to just exctrude.

xblax, 2024-03-12:
I think the "extrude to change filament" approach more comfortable than the "conventional way". But it is also possible to manually pull the filament if the extruder stepper is disabled.

Regarding the KlipperScreen Macros: I didn't have KlipperScreen in mind when I wrote the LOAD_FILAMENT macro. Probably that must also be reworked to have the expected parameters.

If necessary we can define a different macro for usage in Mainsail / Fluidd.

consp, 2024-03-13:
> I didn't have KlipperScreen in mind when I wrote the LOAD_FILAMENT macro

Works fine though! (for my single test at least)

Tiwatz, 2024-03-15:
I tried the LOAD_FILAMENT macro today. As soon it starts to extrude the heater turns off. That means if we extrude some length the nozzle cools down too much. If we add a `G4 P10` after the extrude command seems to do the trick.

xblax, 2024-03-15:
@Tiwatz
Ah yes good catch, didn't notice it. But M400 would probably be better. That macro must be reworked to be compatible with KlipperScreeen anyway:
https://klipperscreen.readthedocs.io/en/latest/macros/#load_filament-unload_filament

KlipperScreen does not expect automatic heating.

Probably as interactive version we could define LOAD_MATERIAL instead.

closed 2024-03-24 by xblax

xblax, 2024-03-24:
I fixed the `LOAD_FILAMENT` and `UNLOAD_FILAMENT` macros, when I also implementing the new Macro `LOAD_MATERIAL` for https://github.com/xblax/flashforge_adm5_klipper_mod/issues/16

`LOAD_FILAMENT` / `UNLOAD_FILAMENT` now does not set the extruder temperature, as it's intended by KlipperScreen. Interactive Loading can be done via `LOAD_MATERIAL`.

Did extensive testing and think that unloading is fine, if needed.