xblax/flashforge_ad5m_klipper_mod #62: Possibility to split Klipper install to a Pi

jaimbo, 2024-03-13:
Sorry, not sure if an issue is the right place to raise this but I am wondering if it will be possible to split the Klipper install to the more traditional setup of using a RPi as the main Klipper host and connecting to the onboard MCU in the printer like a traditional install?

Feels like this would allow for more Klipper functionality and avoid the limitations of the onboard hardware?

consp, 2024-03-13:
You'd have to passthough the uart via network or attach the two uarts required to the pi, could be done but the nice thing is a selfcontained package. Maybe look at [klippventure](https://github.com/synthread/Klippventurer)?

xblax, 2024-03-13:
I discussed the various options to support Klipper in the "early" stage, before this mod was eventually created.
https://github.com/g992/flashforge-ad5m-5mpro-research/issues/6

I think solutions with external Klipper would create more complications than flexibility or require in-depth hardware modifications. I would rule out that this mod goes into that direction again. I also don't feel like we really are limited by the hardware, all the relevant features seem to work.

Nice thing about this self-contained solutions it that it basically works for everyone and we can even support all hardware features like the buzzer, display, built-in camera, USB ports etc. In that sense it's also very economical.

However, if you want you can take mod Linux environment as the basis to connect something external, that's would probably be the better choice than the default Linux.

jaimbo, 2024-03-13:
> You'd have to passthough the uart via network or attach the two uarts required to the pi, could be done but the nice thing is a selfcontained package. Maybe look at [klippventurer](https://github.com/synthread/Klippventurer)?

Thanks, I'll take a look at that project, it might be what I'm looking for!

> I discussed the various options to support Klipper in the "early" stage, before this mod was eventually created. [g992/flashforge-ad5m-5mpro-research#6](https://github.com/g992/flashforge-ad5m-5mpro-research/issues/6)
> 
> I think solutions with external Klipper would create more complications than flexibility or require in-depth hardware modifications. I would rule out that this mod goes into that direction again. I also don't feel like we really are limited by the hardware, all the relevant features seem to work.
> 
> Nice thing about this self-contained solutions it that it basically works for everyone and we can even support all hardware features like the buzzer, display, built-in camera, USB ports etc. In that sense it's also very economical.
> 
> However, if you want you can take mod Linux environment as the basis to connect something external, that's would probably be the better choice than the default Linux.

Is it possible to run things like Obico, KAMP etc. on the built in hardware?

xblax, 2024-03-13:
> Is it possible to run things like Obico, KAMP etc. on the built in hardware?

KAMP is just a set of Macros, so I don't see a reason why it could not work. Don't no much about Obico, but if it just connects to Moonraker you can connect it externally via network - but Klipper and Moonraker can run on the printer.

You will have troubles to use existing install scripts unless some creates specific ones for this mod.

xblax, 2024-03-13:
Hope this has been answered. Will close it since it's not really an issue with the features of the mod. But you're welcome to open a [discussion thread](https://github.com/xblax/flashforge_adm5_klipper_mod/discussions) about possibilities to connect and external Raspberry Pi.

closed 2024-03-13 by xblax