xblax/flashforge_ad5m_klipper_mod #63: Octoeveywhere Support

skylatian, 2024-03-13:
One of the reasons I had for running klipper on the AD5m Pro was to use Octoeverywhere for remote connections! I had a feeling it wouldn't work, but gave it a shot anyways. 
I had to run the script with `sh install.sh` and this was the output. Interestingly, it ran in K1/K1 max mode.
[image]

What kind of work would it take to get this running? I'm going to poke around with [this](https://github.com/D3vil-Design/K1-OctoEverywhere) but I have a feeling that it's very similar to the "K1 mode" of the official installer.

consp, 2024-03-13:
You'd have to build a venv environment for it. I guess it's doable depending on what is exactly required. Never used it but the description tells me it's going to eat resources.

```
if grep -Fqs "ID=buildroot" /etc/os-release
then
    IS_K1_OS=1
```

a K1 build would be triggered yes.

If you want to try building it on the device: `pip3 install virtualenv` and try again. Though I caution you that there is no compiler (and buildroot doesn't supply one) on the device so any cpython modules will tail to build. You'd have to cross compile them and install them that way.

mallison01, 2024-03-16:
I may try this, I really want obico since they offer a self hosted server which I run.

They do offer a container for obico but it fails to run and I have not had time to try further.

I may look into octoeverywhere on a VM or container if someone has one.

I will look into running this externally if possible and report back if I figure it out.

I'm fine with offloading stuff to an external device.

qfertig, 2024-03-18:
> I may try this, I really want obico since they offer a self hosted server which I run.
>
> They do offer a container for obico but it fails to run and I have not had time to try further.
>
> I may look [...]

Yo! have you tried this? I have an extra PI and an old laptop both sitting around that could probably run the companion app, for octoprint, or obico. Assuming it will interact just fine, I will probably try when I get off work, but just figured I would shoot and see if you have tried it yet before I bothered.

consp, 2024-03-18:
Looking at the requirements further (and the install file), you'd have to install it manually and create a venv with precompiled wheels. The install.sh script not going work due to it assuming it's on a debian based system or the K1. Neither is going to be compatible.

Looking at the requirements:
`websocket_client` -> should work
`requests` -> should work
`octoflatbuffers` -> might work
`pillow` -> needs to be a wheel but is possible, cannot be installed on the device so you need to compile the wheels beforehand
`certifi` -> should work
`rsa` -> should work, but why... will be slow anyways <sub>(and also, _NEVER_ claim "secure" when using this library anywhere and it's only used for security theatre anyways since the threat vector is a man-on-the-device with root access, when that happens everything done doesn't matter anymore and is pointless) </sub>
`dnspython` -> should work
`httpx` -> should work, optional-dependencies might not work due to required external libraries
`urllib3` -> should work, same issues as httpx
`sentry-sdk` -> ... (remove if possible)
`configparser` -> should work
`paho-mqtt` -> not required and won't work due to missing dependencies.

Haven't looked further at what it does exactly but if it's just a simple tls tunnel it should be doable if someone wants to put the time into it to create the venv. If it starts modifying data it might eat up a lot of memory.

If it starts re-encoding streams or making screenshots it's unlikely to work due to memory constraints.

skylatian, 2024-03-19:
### It worked for me as well, but *without* config changes!

I also finally got time to give Octoeverywhere Companion a shot, and it works great — I didn't need to run a local Moonraker instance or change any config files (detail below). I'm totally fine running this externally, though down the line it would be great to have OctoEverywhere running on the AD5M.

I'm running this on an Ubuntu VM on a Synology NAS, but the process should be basically the same on any Ubuntu or Debian machine. This is basically the same as the steps [here](https://octoeverywhere.com/getstarted?source=blog_companion&companion=t), just documenting them here for reference.

## Adventurer 5M (Pro) OctoEverywhere Companion Install

On the companion device:

`sudo apt update`
`sudo apt install curl` (may be required, it was for my barebones Ubuntu image)
`bash <(curl -s https://octoeverywhere.com/companion.sh)`

While following the setup prompts that appeared, Companion failed to find a Moonraker instance at the port I provided. However, the installer then prompted for the full IP+UI port of the machine. I chose Fluidd, and it detected the Moonraker install and worked right away!

[image]

skylatian, 2024-03-19:
Ah, gotcha! Re-reading I can see that you meant the config of the local Moonraker install. Pointing it to Fluidd worked for me, but only after the initial Moonraker port connection failed.

Could you clarify what you mean by "just installed it on the Octoprint?" Do you mean that you installed it on the device you're running Octoeverwhere Companion on?

consp, 2024-04-16:
I think the guide from @skylatian is probably best for now, I've copied it to the docs in docs/OCTO_EVERYWHERE.md with a reference to this thread and the OP. See e11fbc0

AJolly, 2024-05-11:
Anyone have any luck getting the obico plugin to connect? (obviously, youd need to use the cloud instance or run your own obico server elsewhere)

mallison01, 2024-05-12:
I have it running on another raspberry pi and tweaked the obico cfg to point to the AD5M. I have a self hosted obico server. 

The only issue is sometimes I have to reboot the pi after I turn on the AD5M for everything to connect promptly. Otherwise it works. 

I am out of town right now but I can share the setup later when I'm back. 

But I think it's setup like someone else did with octoeverywhere

AJolly, 2024-05-12:
Yeah I'd be interested if you have a guide.  I couldn't quite get it to work, first there wasn't bash then no apt-get admittedly didn't try further.

vbence121, 2024-07-01:
> Yeah I forgot to mention this requires no changes on the 5M itself. I couldn't get past the Moonraker part so just installed it on the Octoprint. Thanks for the info on running straight off ubuntu and on the synology.

Just to clarify. By "no changes on the 5M itself" you mean stock firmware or with the moded klipper firmware?

skylatian, 2024-07-01:
@vbence121 No changes to the modded firmware, which exposes moonraker. As far as I know, the original firmware does not use moonraker, meaning obico and octoeverywhere can’t control the printer

consp, 2024-07-08:
> As far as I know, the original firmware does not use moonraker, meaning obico and octoeverywhere can’t control the printer.

To be precise, the OEM firmware runs klippy and firmwareExe, which is an all-seeing-all-controlling blob containing the web/api/gui/external interface component. There is no way to interface with that which is (one of) the reason(s) for the mod (and why I use it).

con80, 2025-06-17:
Are you allowed to make a vm a companion?

closed 2025-09-16 by xblax