xblax/flashforge_ad5m_klipper_mod #65: klipperscreen: camera panel does not load, crashes after error message

xblax, 2024-03-13:
Tested the 'Camera' function of KlipperScreen. It fails to load on shows an error, on screen and in the log. After pressing the button 'Back' it Klipper Screen seems to hang. The screen just stays black.

Might be fixed by enabling libmpv in buildroot.

```
2024-03-13 21:37:48,447 [screen.py:show_error_modal()] - Showing error modal: Unable to load panel camera Cannot find libmpv in the usual places. Depending on your distro, you may try installing an mpv-devel or mpv-libs package. If you have libmpv around but this script can't find it, consult the documentation for ctypes.util.find_library which this script uses to look up the library filename.

Traceback (most recent call last):
  File "/root/printer_software/KlipperScreen/screen.py", line 319, in show_panel
    self.panels[panel_name] = self._load_panel(panel).Panel(self, title, **kwargs)
                              ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/KlipperScreen/screen.py", line 306, in _load_panel
    return import_module(f"panels.{panel}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/root/printer_software/KlipperScreen/panels/camera.py", line 1, in <module>
    import mpv
  File "/root/printer_software/KlipperScreen/KlipperScreen-env/lib/python3.11/site-packages/mpv.py", line 52, in <module>
    raise OSError("Cannot find libmpv in the usual places. Depending on your distro, you may try installing an mpv-devel or mpv-libs package. If you have libmpv around but this script can't find it, consult the documentation for ctypes.util.find_library which this script uses to look up the library filename.")
OSError: Cannot find libmpv in the usual places. Depending on your distro, you may try installing an mpv-devel or mpv-libs package. If you have libmpv around but this script can't find it, consult the documentation for ctypes.util.find_library which this script uses to look up the library filename.

2024-03-13 21:37:48,782 [functions.py:get_software_version()] - Error getting git version: b'fatal: not a git repository (or any of the parent directories): .git\n'
2024-03-13 21:37:50,259 [KlippyGtk.py:Dialog()] - Showing dialog Error (width=800, height=480)
```

consp, 2024-03-13:
Good one, don't have a camera so never tested it properly. Including libmpv would probably fix it, shoud't be too much of an issue.

xblax, 2024-03-13:
Btw. most webcams supported by Linux should work, if you have any. Also don't have the original camera yet.

consp, 2024-03-14:
Good point, should have one somewhere, ancient logitech thing so should work. Pro should arrive in a few days, maybe I'll just wait for that.

I'll also run a ldd on all the compiled modules in klipperscreen to check what it uses, just to be sure.

the-redone, 2024-03-14:
Was going to report this last night, but thought someone might have already :)

Tried with a camera plugged in and removed, same error both times 

Selecting the info icon on the main menu (bottom left corner) also crashes klipper, can't remember the error, but it does restart

consp, 2024-03-14:
> Tried with a camera plugged in and removed, same error both times

That's the be expected though. The library which is missing gets loaded quite early. I'll check if it's the simple lib error tonight (need to rebuild anyway) and see if that fixes it.

>  firmware restart gets it working again, klipper restart ends in another error.
if you know how to ssh to the machine (see docks for login credentials) execute:

```
/etc/init.d/S80klipperscreen stop
/etc/init.d/S80klipperscreen start
```
And it should restart. Restarting klipper itself has not effect since that's a different program (restarting klipper does not restart klipperscreen)

the-redone, 2024-03-14:
> > Tried with a camera plugged in and removed, same error both times
> 
> That's the be expected though. The library which is missing gets loaded quite early. I'll check if it's the simple missing lib error by adding it tonight (need to rebuild anyway due to the mismatch in setsid) and see if that fixes it.
> 
> > firmware restart gets it working again, klipper restart ends in another error.
> 
> if you know how to ssh to the machine (see docks for login credentials) execute:
> 
> ```
> /etc/init.d/S80klipperscreen stop
> /etc/init.d/S80klipperscreen start
> ```
> 
> And it should restart. Restarting klipper itself has not effect since that's a different program (restarting klipper does not restart klipperscreen)
> 
> There is no simple "patch" so for now try avoid the camera panel.

Ye sorry the last part of my message was about the main screen when klipperscreen has loaded, there's a red info button on the main menu, if you press it it takes me here and only a firmware restart gets it working again.

[image]

consp, 2024-03-14:
Just edited the message; That's the emergency abort, it kills everything. Same as in mainsail/fluidd.

consp, 2024-03-14:
Note: this pulls in ffmpeg. The tarball will get a LOT bigger

consp, 2024-03-14:
It works ... but ... at about 0.000000001 FPS. still waiting for the first frame to render.

...

I can report the OOM deamon works as expected.

consp, 2024-03-14:
I guess the best option is to disable the screen. Unless klipperscreen can manage some DMA magic and render directly to screen this is not going to happen.

xblax, 2024-03-14:
Yes, just disable it then. I guess showing the camera on the printer screen is a bit pointless anyway.

consp, 2024-03-14:
The good thing coming from this excersise is the proper test of the OOM killer I guess. Since Klipperscreen is the lowest (highest niceness) it gets killed as it should.

consp, 2024-03-18:
"Fixed" in https://github.com/KlipperScreen/KlipperScreen/commit/f6af6e5bbdc97a97c7a050982b9e240b59a4623b since it will crash klipper (and maybe the system) due tot memory overusage forcing the OOM to kick in.