xblax/flashforge_ad5m_klipper_mod #73: Auto Bed Leveling inaccurate

mBlinkii, 2024-03-15:
Bed levelling inaccurate due to filament residue on the nozzle. the preparation is not the same as with the original software, i think it is better solved there so that no filament remains on the nozzle.

this is how the FF software does it.

1. home
2. heat up the bed
3. move X to the left to the center, bring the bed close to the nozzle.
4. heat up the nozzle
5. lower the nozzle onto the bed
6. raise the nozzle quickly and move it slightly to the right
7. lower the nozzle and wipe it off
8. cool the nozzle
9. quickly lift the nozzle and remove the remaining filament
10. move the nozzle to the first test position

xblax, 2024-03-15:
@mBlinkii 
Can you tell what exactly is different and where you think issues come from? The Macros do a similar routine via AUTO_BED_LEVEL, I had not really any issues with Filament on the Nozzle. First layers are more or less looking pristine.

Actually on stock the probing itself is terribly inaccurate, because probing is done to fast, resulting in big variance between repetitions ( I measured that ).

mBlinkii, 2024-03-15:
> @mBlinkii Can you tell what exactly is different and where you think issues come from? The Macros do a similar routine via AUTO_BED_LEVEL, I had not really any issues with Filament on the Nozzle. [...]

the first layer is too high with this mod, with the original firmware the first layer is perfect. i noticed during the auto leveling that there is still filament left on the nozzle tip. i just thought that this might be the reason why the first layer starts too high.

xblax, 2024-03-15:
> the first layer is too high with this mod, with the original firmware the first layer is perfect. i noticed during the auto leveling that there is still filament left on the nozzle tip. i just thought that this might be the reason why the first layer starts too high.

You do the leveling via the AUTO_BED_LEVEL macro with the correct filament temperature? 
Sure if there is filament on the Nozzle during leveling then the results would be off, but for me that has rarely be the case, so far. I'm open to improving it, but have to understand what exactly makes a difference.

There are also other issues at play, i.e. wrong extrusion distance resulting in wrong flow-rate: https://github.com/xblax/flashforge_adm5_klipper_mod/discussions/70

mBlinkii, 2024-03-15:
> You do the leveling via the AUTO_BED_LEVEL macro with the correct filament temperature? Sure if there is filament on the Nozzle during leveling then the results would be off, but for me that has [...]

I did it with 55°C Bed and 220°C Nozzle, like the FF Software. i also repeated it several times, but each time there was filament residue on the nozzle.

I also tested it again with the FF software and I found its cleaning the nozzle better.

thanks for the tip with the flow rate, i will test it tomorrow.

Thanks for the great work.

mBlinkii, 2024-03-16:
I experimented a bit with the AUTO_BED_LEVEL macro and with these changes the first layer looks much better to me. 

[image]

The unmodified macro on the left and the macro I modified on the right.

`[gcode_macro AUTO_BED_LEVEL]
gcode:
    {% set bed_temp = params.BED_TEMP|default(60)|float %}
    {% set extruder_temp = params.EXTRUDER_TEMP|default(200)|float %}
    {% set profile = params.PROFILE|default("auto") %}

    # clear old mesh and home if necessary
    BED_MESH_CLEAR
    {% if not "xyz" in printer.toolhead.homed_axes %}
        G28
    {% endif %}

    # bed heating
    G90                     ; absolute coordinates
    G1 Z10 F1800             ; raise bed
    M140 S{bed_temp}        ; set bed temp
    # wait for bed to reach temp, don't wait for stabilization
    TEMPERATURE_WAIT SENSOR=heater_bed MINIMUM={bed_temp}
    LOAD_CELL_TARE          ; reset tare before manual probe

    # nozzle heating
    #PROBE                   ; moove nozzle to bed
    G91                     ; relative position
    G1 Z-0.25               ; release pressure from bed
    M104 S{extruder_temp}   ; set nozzle temperature
    # wait for nozzle to reach temp, don't wait for stabilization
    TEMPERATURE_WAIT SENSOR=extruder MINIMUM={extruder_temp}

    # get into position
    G90                     ; absolute coordinates
    G1 X-20 Y110 Z5 F18000  ; move in position for nozzle wipe
    PROBE                   ; moove nozzle to bed
    G91                     ; relative position
    G1 X-10 F900            ; remove filament
    G1 X20 Z10 F18000       ; lift nozzle and move right
    PROBE 
    G91                     ; relative position
    G1 Z0.1 F300            ; release some pressure from bed

    # wipe move to right
    G1 X60 F900             ; wipe move to right
    G1 Y-1 F900
    # wipe move to left
    G1 X-30 F900            ; wipe move to left
    G1 Y-1 F900

    # wait for nozzle cooldown to 120°
    G1 Z-0.1                ; put full pressure on bed
    M106 S255               ; turn on fan
    M104 S0                 ; turn of noozle heating
    TEMPERATURE_WAIT SENSOR=extruder MAXIMUM=120
    M107                    ; turn off fan
    G1 X10 F300             ; move to remove cold filament
    G1 Z10 F18000           ; lift nozzle to remove filament
    

    # start the calibration
    BED_MESH_CALIBRATE PROFILE={profile}`

mBlinkii, 2024-03-16:
It's not perfect yet, I also compared the mesh from the FF software and from this mod again, there are only minimal differences. I think FF set a Z offset when printing starts.

consp, 2024-03-16:
> It's not perfect yet, I also compared the mesh from the FF software and from this mod again, there are only minimal differences. I think FF set a Z offset when printing starts.

`SET_GCODE_OFFSET Z=` is triggered by the ff software in printFile and printSerial so I guess you are right.

mBlinkii, 2024-03-16:
is it possible to get the value? can we debug the flashforge ui? I opened the firmwareExe with a hex tool, but couldn't find anything useful.

consp, 2024-03-16:
> is it possible to get the value? can we debug the flashforge ui? I opened the firmwareExe with a hex tool, but couldn't find anything useful.

It's read from a config file. My guess would be the -0.25 already known.

mBlinkii, 2024-03-18:
I change the offset in the START_PRINT macro from 0.0 to -0.8, this was perfect for me - 0.25 was to much. I think everyone has to adapt this themselves so that the first layer is perfect. in the Adventurer5M.json from my printer h "zLevelOffset" : -1.8200000524520874, dont know how ff calculates this value.

consp, 2024-03-18:
Mine is -0.82999998331069946.

It's set during doAutoCalibrateMesh, so my guess would be during autoleveling.

xblax, 2024-03-20:
> The macro does a G28 on start. It’s working for me now, ff sets a z offset before printing starts. You need to add / change the z offset in the Start Print macro. I used -0,08. this was fine for me.

@mBlinkii 
I think the proper way would be tun run PROBE_CALIBRATE to in the middle of the bed and and adjust the probe z-offset. See: https://www.klipper3d.org/Probe_Calibrate.html#calibrating-probe-z-offset

For me the default -0.25 were pretty accurate.

Can you also check your extruder rotational distance? See https://github.com/xblax/flashforge_adm5_klipper_mod/discussions/70
I need a few more calibration results before changing the default.

xblax, 2024-03-21:
@mBlinkii 
I looked at your Auto Leveling Macro. I noticed you disabled the probe/move to bed before heating the nozzle. Do you think it's better that way because molten filament can "drip" off the nozzle? My initial thought was to move to bed to prevent oozing, but maybe that's not optimal.

You also added a second probe after initial wipe move and a move after cooling the noozle. That also make sense, but I think it can be simplified and improved a bit. Because another issue I noticed is that on the wipe move in the opposite direction, the nozzle tends to pick up the filament it just wiped off.

This version avoids it, by doing two wipe moves on a single line. Wipe from right towards center, then move and wipe from left towards center. It also puts a bit more pressure onto the bed for the Wipe moves than the old version:

```
[gcode_macro AUTO_BED_LEVEL]
gcode:
    {% set bed_temp = params.BED_TEMP|default(60)|float %}
    {% set extruder_temp = params.EXTRUDER_TEMP|default(200)|float %}
    {% set profile = params.PROFILE|default("auto") %}

    # clear old mesh and home if necessary
    BED_MESH_CLEAR
    {% if not "xyz" in printer.toolhead.homed_axes %}
        G28
    {% endif %}

    # bed heating
    G90                     ; absolute coordinates
    G1 Z10 F1800            ; raise bed
    M140 S{bed_temp}        ; set bed temp
    # wait for bed to reach temp, don't wait for stabilization
    TEMPERATURE_WAIT SENSOR=heater_bed MINIMUM={bed_temp}

    # nozzle heating
    G91                     ; relative position
    G1 Z-0.25               ; release pressure from bed
    M104 S{extruder_temp}   ; set nozzle temperature
    # wait for nozzle to reach temp, don't wait for stabilization
    TEMPERATURE_WAIT SENSOR=extruder MINIMUM={extruder_temp}
    LOAD_CELL_TARE          ; reset tare before manual probe

    # wipe to left
    G90                     ; absolute coordinates
    G1 X50 Y110 Z5 F18000   ; get into position
    PROBE                   ; moove nozzle to bed
    G91                     ; relative position
    G1 X-40 F900            ; wipe move left

    # wipe to right
    G90                     ; absolute coordinates
    G1 X-50 Y110 Z5 F18000  ; get into position
    PROBE                   ; moove nozzle to bed
    G91                     ; relative position
    G1 X40 F900             ; wipe move right

    # wait for nozzle cooldown to 120°
    M106 S255               ; turn on fan
    M104 S0                 ; turn of noozle heating
    TEMPERATURE_WAIT SENSOR=extruder MAXIMUM=120
    M107                    ; turn off fan
    # move to remove cold filament
    G1 X10 F300
    
    # start the calibration
    BED_MESH_CALIBRATE PROFILE={profile}
```

Can you give this version a try?

mBlinkii, 2024-03-21:
> @mBlinkii I looked at your Auto Leveling Macro. I noticed you disabled the probe/move to bed before heating the nozzle. Do you think it's better that way because molten filament can "drip" off the nozzle? My initial thought was to move to bed to prevent oozing, but maybe that's not optimal.
> 
> You also added a second probe after initial wipe move and a move after cooling the noozle. That also make sense, but I think it can be simplified and improved a bit. Because another issue I noticed is that on the wipe move in the opposite direction, the nozzle tends to pick up the filament it just wiped off.
> 
> This version avoids it, by doing two wipe moves on a single line. Wipe from right towards center, then move and wipe from left towards center. It also puts a bit more pressure onto the bed for the Wipe moves than the old version:
>
> Can you give this version a try?

Hi, thanks for your answers. I will test it and report back here. 

With the macro, I wanted to avoid the knob at the home point on the bed. 

The 2nd lift and move to the right was to avoid pulling the filament from the first time and the movement after cooling to get rid of any filament residue.

mBlinkii, 2024-03-21:
I haven't checked the rotation distance yet, I'll check it and let you know.

mBlinkii, 2024-03-21:
@xblax, i did PROBE_CALIBRATE and im getting -0.392, also the new Level macro works great. i will do now a test print.

mBlinkii, 2024-03-21:
Something went wrong for me, with this new offset, I add it under the probe section, the first layer is also too high with this offset, will check it tomorrow.

xblax, 2024-03-21:
> Something went wrong for me, with this new offset, I add it under the probe section, the first layer is also too high with this offset, will check it tomorrow.

I'm not surprised that -0.392 is too far off the bed. How do you do the calibration?

I do this right after AUTO_BED_LEVEL, I move the print head to (0,0) and do the PROBE_CALIBRATE. The bed is still heated and the load cell is tared already. Otherwise it would need manual LOAD_CELL_TARE before doing PROBE_CALIBRATE.

The calibration is done via paper test. Move the bed up until the paper is "stuck" between the bed and noozle. The bed moves down 5mm after probing, so maybe start with raising it 4.8mm. Then release the pressure until the paper can just move without much friction. That gives me an offset between of roughly -0.25. There is a variance of about +-0.025mm where it's hard to determine what is the right amount of friction. That must be fine tuned anyway and might also vary depending on the bed surface (textured vs. smooth).

mBlinkii, 2024-03-22:
> I'm not surprised that -0.392 is too far off the bed. How do you do the calibration?
>
> I do this right after AUTO_BED_LEVEL, I move the print head to (0,0) and do the PROBE_CALIBRATE. The bed is [...]

I did it before Auto level. So this was my mistake. I will redo this today after work. Thanks for this tip.

consp, 2024-03-22:
> There is a variance of about +-0.025mm where it's hard to determine what is the right amount of friction.

Isn't that simply the imperfections in paper density anyway?

xblax, 2024-03-22:
> Isn't that simply the imperfections in paper density anyway?

I think paper thickness is quite uniform around 0.1mm for 80g paper. It's just not easy to determine the *exact* same amount of friction every time. Anyway it's not that important, since the probe offset should not need to be changed often after dialed in once.

I recommend the Klipper docs, they are quite good on that topic: https://www.klipper3d.org/Bed_Level.html#the-paper-test

> After successfully performing bed leveling, one may go on to calculate a more precise value for the combined impact of "thermal expansion", "thickness of the paper", and "amount of friction felt during the paper test".
>
> This type of calculation is generally not needed as most users find the simple "paper test" provides good results.

In our case we would slightly adjust the measured probe offset. Or set a GCode offset in the Slicer. I can try to do test prints on smooth and textured print bed with the adjusted rotational distance for the extruder.

My general tendency is to add in the improved macro and leave the default probe offset at -0.25 as is.

mBlinkii, 2024-03-22:
So I have now done it as described, first Auto Bed Level and then in the heated state once PROBE_CALIBRATE, my value is now -0.175. The first layer is now perfect.

I have also tested it with the standard value of -0.25 and I can only say that both work perfectly for me with the new level macro.

mBlinkii, 2024-03-22:
so the value -0.25 should be optimal for everyone.

consp, 2024-03-22:
Just tried the auto level script with a PC sheet and that also works as expected. Just enough space to get it working.

xblax, 2024-03-23:
Added the improved Macro on master. After doing some more test prints and taking the 0.1 thickness of the paper into account, I actually start to think the z-offset can be lowered a bit (closer to bed).

But will have to do more prints to come to a final conclusion. Will keep it at -0.25 for now.

gpena208777, 2024-03-30:
I have noticed this as well even with the newer Beta. 
What's interesting is the Klipper mod also starts bed leveling without heating the bed.

consp, 2024-04-01:
> What's interesting is the Klipper mod also starts bed leveling without heating the bed.

That only happens when you use BED_MESH_CALIBRATE instead of AUTO_BED_LEVEL

AJolly, 2024-04-19:
some of this macro doesn't make sense for example
```
G91                     ; relative position
G1 Z-0.25               ; release pressure from bed
```
This would drive the nozzle into the bed. Which doesn't really matter because before that we raised the nozzle to Z10 anyways.

Heres what I'm using.  

```
[gcode_macro AUTO_BED_LEVEL]
gcode:
    {% set bed_temp = params.BED_TEMP|default(60)|float %}
    {% set extruder_temp = params.EXTRUDER_TEMP|default(200)|float %}
    {% set profile = params.PROFILE|default("default") %}

    M118 clear old mesh and home if necessary 1/8
    BED_MESH_CLEAR
    {% if not "xyz" in printer.toolhead.homed_axes %}
        G28
    {% endif %}

    M118 bed heating 2/8
    G90                     ; absolute coordinates
    G1 Z10 F1800            ; raise bed
    M140 S{bed_temp}        ; set bed temp
    M104 S{extruder_temp}   ; set nozzle temperature so we dont need to wait
    M118 wait for bed to reach temp, don't wait for stabilization
    TEMPERATURE_WAIT SENSOR=heater_bed MINIMUM={bed_temp}
    
    M118 nozzle heating 3/8
    TEMPERATURE_WAIT SENSOR=extruder MINIMUM={extruder_temp}

    M118 wipe to left 4/8
    G90                     ; absolute coordinates
    G1 X50 Y110 Z5 F18000   ; get into position
    M400
    M118 reset  tare 
    LOAD_CELL_TARE          ; reset tare before manual probe
    PROBE                   ; move nozzle to bed
    G91                     ; relative position
    G1 X-40 F900            ; wipe move left

    M118 wipe to right 5/8
    G90                     ; absolute coordinates
    G1 X-50 Y110 Z5 F18000  ; get into position
    M400
    M118 reset tare again
    LOAD_CELL_TARE          ; reset tare before manual probe
    PROBE                   ; move nozzle to bed
    G91                     ; relative position
    G1 X40 F900             ; wipe move right

    M118 wait for nozzle cooldown to 120° 6/8
    M106 S255               ; turn on fan
    M104 S0                 ; turn off nozzle heating
    
    M107                    ; turn off fan
    TEMPERATURE_WAIT SENSOR=extruder MAXIMUM=120
    
    M118 move to remove cold filament 7/8
    G1 X10 F300

    M118 start the calibration 8/8
    BED_MESH_CALIBRATE PROFILE={profile}
```
I also changed the probe config to include retries:

```
[probe]
pin: !PB4
speed: 1
lift_speed: 8
samples: 4
samples_tolerance: 0.02
samples_tolerance_retries: 3
```

4:16 PM
probe at -105.000,-105.000 is z=-1.752500
4:16 PM
probe at -105.000,-105.000 is z=-1.745000
4:16 PM
probe at -105.000,-105.000 is z=-1.750000
4:16 PM
probe at -105.000,-105.000 is z=-1.740000
4:16 PM
Probe samples exceed tolerance. Retrying...
4:16 PM
probe at -105.000,-105.000 is z=-1.745000
4:16 PM
probe at -105.000,-105.000 is z=-1.737500
4:16 PM
probe at -105.000,-105.000 is z=-1.720000

GAMRABADR, 2024-09-15:
a me non va ma ora non sembra nemmeno laltezza piatto ma lestrusione esce come se dovessi cagare molle ci mette un ora e quindi la prima parte non la fa nemmeno penso che bisogno flaschare e ritornare al frimware originale mannaggia a me andava tutto prima stampa senza guardare clicavo stampa e andavo a dormire ora non va piu un cazzo tolgo tutto se si va avanti di sto passo si creano danni alla macchina a forza di pasticciare se ci fosse gente capace a programmare sarebbe gia uscito il primo giorno la macchina e diventata inutilizzabile

GAMRABADR, 2024-09-15:
piatto giallo e piatto nero hanno due spessori diversi fralaltro

GAMRABADR, 2024-09-15:
come si fa a farli fare le line di spurgo prima della stampa ! sono Noob e nuovo

GAMRABADR, 2024-09-15:
[gcode_macro START_PRINT]
gcode:
    {% set bed_temp = params.BED_TEMP|default(60)|float %}             ; bed temp, usually set by slicer
    {% set extruder_temp = params.EXTRUDER_TEMP|default(200)|float %}  ; extruder temp, usually set by slicer
    {% set disable_priming = params.DISABLE_PRIMING|default(false) %}  ; allow disabling of priming before print
    {% set disable_skew = params.DISABLE_SKEW_CORRECT|default(false) %}; if true it will not apply the skew correction profile
    {% set force_leveling = params.FORCE_LEVELING|default(true) %}    ; if true it forces the bed level process

qui giusto ?

loss-and-quick, 2025-09-19:
Checkout my variant of cleaning nozzle: https://github.com/loss-and-quick/flashforge_ad5m_klipper_mod/blob/stm-dev/printer_configs/common/macros.cfg#L79