xblax/flashforge_ad5m_klipper_mod #75: klipperscreen: Timezone settings to show the correct time on the clock

consp, 2024-03-16:
See #74

xblax, 2024-03-17:
Timezone data should be enabled in Buildroot with default UTC. Also for the base variant then to keep it consistent, not only klipperscreen.

I guess it could be configured via the user overlay - maybe we can also make the setting persistent.

consp, 2024-03-17:
> I guess it could be configured via the user overlay - maybe we can also make the setting persistent.

You can set TZ but that will have no effect on the clock afaik without a form of time control daemon. UTC is indeed the default but that's since nothing has been set (always defaults to utc since that's what the fake RTC runs at).

xblax, 2024-03-17:
Afaik timezone is supported by the Kernel and just needs some user-space initialization to set the correct time zone after boot. I think the configuration is usually just manged by a symlink to the correct timezone file. 

The fake-hwcloc should be kept in UTC (not in local time).

consp, 2024-03-17:
> The fake-hwcloc should be kept in UTC (not in local time).

True, but I was expecting date to respect TZ info. But since the TZ_INFO and the TZDATA packages is not installed it won't. I'll add them and see it helps, could add a symlink in the user config to a symlink in /usr/share/zoneinfo which again symlinks to /etc/localtime. Changing the symlink in the user config dir will alter the timezone in that way.

consp, 2024-03-18:
Including TZ_INFO fixes the system ignoring the timezone.

closed 2024-03-18 by consp

xblax, 2024-03-19:
Looks good. Pushed this change to the master branches.