xblax/flashforge_ad5m_klipper_mod #83: When is the pcb_fan supposed to start running?

function3d, 2024-03-19:
I have watched the fan and it does not start when printing starts, I think the pcb_fan should be a motherboard_fan in the klipper configuration instead of a generic_fan, since as generic_fan does not seem to be started.

Or maybe there is some control by temperature that I don't know about?

function3d, 2024-03-19:
Ok I have seen it now, the pcb_fan starts on the START_PRINT command, the problem was that I used my own START_PRINT command, in any case I think it is better to set the pcb_fan as motherboard_fan this way you avoid the possibility that it will not start if someone forgets to print without the START_PRINT command

consp, 2024-03-19:
Afaik there is not "standard" for this. Neither klipper docs nor any example configs mention it and the default START_PRINT macro doesn't either. It's something people should be weary about but you can always change it yourself.

xblax, 2024-03-19:
I looked that the Klipper docs, and there is indeed the concept of a controller fan:
https://www.klipper3d.org/Config_Reference.html#controller_fan

> Controller cooling fan (one may define any number of sections with a "controller_fan" prefix). A "controller fan" is a fan that will be enabled whenever its associated heater or its associated stepper driver is active. The fan will stop whenever an idle_timeout is reached to ensure no overheating will occur after deactivating a watched component.

So we could configure the PCB fan to run whenever the steppers are active. Maybe that's a good idea, will have to look into it.

consp, 2024-03-19:
Maybe it's better to rename the pcb_fan to something else since it's neither a pcb cooler not a motherboard cooler (as it's only cooling the stepper drivers anyway).

function3d, 2024-03-20:
Controller_fan allows to specify the heaters and steppers that are associated to the fan, in this case it can be used only when using the steppers x,y,z and not for the extruder when for example only loading or unloading filament, in my case I leave it for all because I have the power supply fan in series with the pcb_fan so when the print is done the printer is completely silent when the hotend cools down

xblax, 2024-03-20:
@function3d 
Yes I will change it to be activated when x,y,z steppers are active. That makes more sense than the current implementation. Will also rename it to stepper_driver_fan. 

I think it would also be nice to test, if a lower then max speed would be sufficient to cool the steppers. Do you run it at 100%?
I have started a similar discussion about the hotend fan, which I think can run at lower speed be turned off at higher temperature: https://github.com/xblax/flashforge_adm5_klipper_mod/discussions/71

Putting the PSU fan in series is also nice idea. I changed my PSU to a genuine Meanwell, now the fan also stays off most of the time and the printer is 100% quiet when not used.

function3d, 2024-03-20:
Sorry I meant in parallel not in series with the power supply fan.

Yes, I run it at 100%, in my case to be in parallel with the power supply I did not think at any time to slow it down because I do not know if it will be problematic for the power supply, but perhaps at 80 or 90% it would cool enough and although the noise reduction is not much would lengthen the life of the fans. I will test the 80%

xblax, 2024-03-21:
I did check the airflow and 80% seems to be a good trade-off between cooling and noise for the stepper driver fan. Probably also for the PSU fan? Without fan the stepper drivers get to ~50°, with fan they are around 30° (not much difference between 80 and 100%). Measured with simple infrared thermometer.

```
[controller_fan stepper_driver_fan]
pin:PB7
stepper: stepper_x, stepper_y, stepper_z
fan_speed: 0.8
idle_timeout: 0
```

closed 2024-03-23 by xblax

xblax, 2024-03-23:
Added this on master now.