xblax/flashforge_ad5m_klipper_mod #86: Camera Stopped Working

pelegw, 2024-03-20:
Hi Everyone, I wonder if anyone ran into this issue as well.
I have a logitech c270 attached to the printer, and on first load it worked well. After I shut the printer down though, it won't load anymore. Tried rebooting the system but still nothing.

When I attach it this is the message I'm getting:

> Mar 20 22:56:08 ADM5 kern.info kernel: [   44.053735] usb 3-1.3: new high-speed USB device number 4 using sunxi-ehci
> Mar 20 22:56:08 ADM5 kern.info kernel: [   44.418450] uvcvideo: Found UVC 1.00 device <unnamed> (046d:0825)
> Mar 20 22:56:08 ADM5 kern.warn kernel: [   44.532636] uvcvideo 3-1.3:1.0: Entity type for entity Extension 4 was not initialized!
> Mar 20 22:56:08 ADM5 kern.warn kernel: [   44.532655] uvcvideo 3-1.3:1.0: Entity type for entity Extension 6 was not initialized!
> Mar 20 22:56:08 ADM5 kern.warn kernel: [   44.532665] uvcvideo 3-1.3:1.0: Entity type for entity Extension 7 was not initialized!
> Mar 20 22:56:08 ADM5 kern.warn kernel: [   44.532675] uvcvideo 3-1.3:1.0: Entity type for entity Processing 2 was not initialized!
> Mar 20 22:56:08 ADM5 kern.warn kernel: [   44.532685] uvcvideo 3-1.3:1.0: Entity type for entity Extension 3 was not initialized!
> Mar 20 22:56:08 ADM5 kern.warn kernel: [   44.532694] uvcvideo 3-1.3:1.0: Entity type for entity Camera 1 was not initialized!
> Mar 20 22:56:08 ADM5 kern.info kernel: [   44.533909] input: UVC Camera (046d:0825) as /devices/platform/soc@3000000/4200000.ehci1-controller/usb3/3-1/3-1.3/3-1.3:1.0/input/input3
> Mar 20 22:56:08 ADM5 user.notice camera-autostart: received ACTION=add /dev/video0
> Mar 20 22:56:09 ADM5 user.notice camera-autostart: started ustreamer for /dev/video0 with res 1280x720

The streamer runs, but I'm getting a no-signal message.

pelegw, 2024-03-20:
Running ustreamer manually gives the following error:
[image]

xblax, 2024-03-21:
Some of the dmesg errors and the first ustreamer error are "normal" I think. I also have them, but the dmesg errors seems to vary a bit depending on the camera model. But the cameras work. Tested the official one an a generic webcam.

I think the issue is that "Mainloop select error()". 
Have you tried full power off? Is there a difference in booting with the camera already attached vs not?

pelegw, 2024-03-21:
The error is because for some reason, the camera is not encoding mjpeg, I have no idea why it worked and stopped working. 
I'll try an Arducam when it arrives next week instead and will update here.

consp, 2024-03-21:
Some camera's have weird quirks, one of my collegues had a camera which after using under linux was tinted blue. Under windows it went normal again but as soon as you used it on _any_ linux machine it went blue again. Some weird register setting in the firmware probably.

pelegw, 2024-03-21:
Launching ustreamer with --io-method=USERPTR seems to solve the problem.

pelegw, 2024-03-21:
Adding the --device-timeout=2 option seems to have solved the issue and the camera is now working. 
This time, it survived multiple printer reboots and camera disconnects. I suggest adding this option to the camear autostart script, it shouldn't affect performance but may help solve this for others.

xblax, 2024-03-22:
@pelegw 
Great you were able to find the root cause! I will look into that parameter and then probably add it to the script.

closed 2024-03-23 by xblax

xblax, 2024-03-23:
Fix is included and master and will be part of the next release.