xblax/flashforge_ad5m_klipper_mod #90: klipperscreen: stops working after change of font size or theme

xblax, 2024-03-23:
After changing these settings in the menu, KlipperScreen just shuts down. No error is shown in the log. After manual restart the settings are applied. I think that might actually be intended behavior, to restart the application for these settings.

We might need a wrapper script to automatically restart KlipperScreen if the process terminates.

I like the large font size setting, I guess no one else tried it yet. Maybe that would even make sense as default.

consp, 2024-03-23:
Good point, I'll have a look. Thought I tested it but maybe that was with running it manually and forgetting about it.

consp, 2024-03-23:
Large also looke better yes

consp, 2024-03-23:
It's a bug.

```
Traceback (most recent call last):
  File "/root/printer_software/KlipperScreen/screen.py", line 13, in <module>
    import gi
ModuleNotFoundError: No module named 'gi'
```
Tries to start with common python3 not the venv, can be fixed in the script

xblax, 2024-03-24:
Added to master and cleaned-up the branch. Can be closed.

closed 2024-03-24 by xblax