xblax/flashforge_ad5m_klipper_mod #93: empty printer.cfg

slydog43, 2024-03-23:
I had .003 working on a A5M fine for a couple of days (loving it) but was trying to get my bed more level and had to power on/off my machine a few times.  I now have an empty printer.cfg somehow.  Don't know if this is a bug or not just want to mention it.

I see that the Klipper is installed screen is getting updated to Klipper is installing is in the works, thanks.   

Great work guys

mBlinkii, 2024-03-23:
i think the file was damaged when the printer was switched off. just paste the content from here https://github.com/xblax/flashforge_adm5_klipper_mod/blob/master/printer_configs/printer.cfg and everything should be fine again.

slydog43, 2024-03-23:
Thanks, will create a backup just incase this happens again.

xblax, 2024-03-24:
Have not observed that yet. Will tag this issue as 'not-reproducible'. If you observe this again, maybe you can provide more information here to help find the root cause.

My assumptions would also be that something like that could be caused by unclean shutdown. But I do it all the time, and didn't notice broken files.

Grea5eMonkey, 2024-03-25:
I had this happen the other day as well after an unclean shutdown. Luckily I hadn't modified the file, so I just re-uploaded it.

consp, 2024-03-25:
Do you know what writes to the files? Moonraker I would guess, maybe it has classic python sync issues, empty files is usually non flushed write buffers.