xblax/flashforge_ad5m_klipper_mod #112: Options in Docs to get Wifi connected: None worked except the ethernet cable + SSH commands option

DougJoseph, 2024-03-29:
Thanks to the amazing wizards that made this mod! I am thrilled with it!

My experience was, and several others have also reported to me, that of all the options in Docs to get Wifi connected: None of them worked except the one "ethernet cable + SSH commands" option. 

https://github.com/xblax/flashforge_adm5_klipper_mod/blob/master/docs/WIFI.md

I think the Klipperscreen option to scan and enter password failed, as apparently the screen's efforts to save the password were prevented. Not sure why.

Also, the "USB klipper_mod folder" method did not work for me, despite very carefully following the directions mulitple times.

This one issue seems to be a hurdle that several are facing. Some are in a situation that precludes them from getting an ethernet cable connected from their router (reportedly).

I should add that my troubles were with v00.03-beta and the klipperscreen variant. I am about to try again on a second AD5M printer, but this time on v00.04. I will report back here, how it goes.

Thanks again for all you do!

Tiwatz, 2024-03-29:
I had the same issue when I installed v00.03. Used the USB option but it did not work. I SSH in to machine and the file was there but it was some mysterious characters in the end of every line. I removed those and then everything worked fine.

DougJoseph, 2024-03-29:
That's good to know. It could be that formatting (perhaps something other than UTF-8 when UTF-8 was needed) caused some USB method failures. I have been well aware of encoding issues for a long time, and I'm fairly confident my .PSK files were made with UTF-8 encoding.

DougJoseph, 2024-03-29:
> I SSH in to machine and the file was there but it was some mysterious characters in the end of every line. I removed those and then everything worked fine.

Did you use some terminal based method for viewing and editing the file? If so, what method? What commands?

consp, 2024-03-29:
> there but it was some mysterious characters in the end of every line

Those are likely windows line endings (^M), you need to convert them to unix format (e.g. with notepad++).

consp, 2024-03-29:
> apparently the screen's efforts to save the password were prevented

If you have the logs (`/var/log/KlipperScreen.log`) they'd be very welcome.

Tiwatz, 2024-03-29:
> 
> Did you use some terminal based method for viewing and editing the file? If so, what method? What commands?

I don't remember exactly but it was vi or nano editor I used in Putty.

DougJoseph, 2024-03-29:
OK, good news. 

Currently I'm installing v00.04-beta (Klipper variant) on a new AD5M. 

I just now videoed testing both the "Klipperscreen UI" method of hand selecting the network SSID and hand typing the password, as well as testing the "USB + klipper_mod folder" method.

For testing the "Klipperscreen UI" method — I again got the red error message at the top. However, unlike last time, I persisted a bit. I ignored the red error message, and waited, and then tapped back into Network, and the printer was connected. Even better, I shut down the printer, rebooted it, and went back into Network, and it was again connected. So, despite the red notice bar claiming there was an error saving the credentials, it seems to have worked. I'm guessing that last time (on the v00.03-beta attempt on the first printer) I did not ignore the red error message, and gave up assuming it did not work. Regrettably, I did not get logs on it, because I had already started the second test by the time I read the request here by @consp to get a log.

For testing the "USB + klipper_mod folder" method: I created the .PSK file, and added it to the USB. I added it as before, with the proper folder structure, to the USB drive. The primary difference this time from last time, is last time had pre-assigned a static IP, and include that IP in the file as shown below:

```
[Settings]
AutoConnect=true
AlwaysRandomizeAddress=false

[Security]
Passphrase=********
 
[IPv4]
Address=192.168.1.28
Netmask=255.255.255.0
Gateway=192.168.1.1
DNS=1.1.1.1 8.8.8.8
``` 

By contrast, this time, I had not pre-assigned a static IP, and this time I did not have that in the file. This time the file's content was:

```
[Settings]
AutoConnect=true
AlwaysRandomizeAddress=false

[Security]
Passphrase=********
 ``` 

Anyhow, this time it worked, and it connected to wifi on its own, right after booting into Klipperscreen!

So, now I have verified the "ethernet cable + SSH commands" method on my first printer, and I have verified both the "Klipperscreen UI" method and the "USB + klipper_mod folder" method on my second printer.
 
I feel sheepish, but perhaps this thread will help others. I'm marking this one as closed, and again I'm so happy with this mod! Thanks again!

closed 2024-03-29 by DougJoseph

consp, 2024-03-29:
Good to hear it worked, considering wifi is always full of strange error methods it's a hit-or-miss scenario. The code for klipperscreen is a bit patchy since it was not designed to be ran with iwd (the wifi manager used in the mod) so it's a work in progress. DIscussed it with the klipperscreen author to merge it and develop it further and they were working on altering the method for dealing with wifi so I'm a bit hesitant to do a lot of work on it, as long as there are no clear -easilly managable- problems.