xblax/flashforge_ad5m_klipper_mod #113: SAVE_CONFIG section 'stepper_z' option 'position_endstop' conflicts with included value

AJolly, 2024-03-29:
Since the position_endstop is in printer.base.cfg, it can't be updated.

This is after running a print, adjusting z via the screen, and attempting to save.

THEGxHOxST, 2024-03-30:
I had to use probe off set, I believe its because the beds endstop is at the bottom.
probe offest can be found in printer.cfg and klipper screen.

xblax, 2024-04-01:
The answer from @THEGxHOxST is correct. Because the bed end stop is at the bottom (and also because the probe is not the end stop), it does not make sense to save the offset to the endstop (it would just lead to accumulated offset error over time).

This was already discussed here: https://github.com/xblax/flashforge_adm5_klipper_mod/discussions/61

I will create a separate issue to add section about bed leveling to the documentation. This should also be explained there.

closed 2024-04-01 by xblax