xblax/flashforge_ad5m_klipper_mod #121: Load Cell Temp

MoonDoggie97, 2024-04-02:
When the printer is in standby doing nothing the load cell temperature is going up to 600+ Celsius is this normal? Will this kill the device quicker

consp, 2024-04-02:
> When the printer is in standby doing nothing the load cell temperature is going up to 600+ Celsius is this normal? Will this kill the device quicker

"Temperature" is just the value of the strain gauges. It's not an actual temperature. 

When it goes idle the bed contracts (due to reducing in temperature). The strain gauge thus change shape and send a signal. This is completely normal and it's zeroed every time you auto-bed-level it (and then it should be heated) so idle will be non-zero when idle and cold.

KathrynHazuka, 2024-04-03:
I came in here about to make the same post and im glad it's been answered. that's curious, and good to know that the cell value isn't actual celsius. I was wondering isn't that going to kill the component too heh. Thank you for answering this for us @consp