xblax/flashforge_ad5m_klipper_mod #124: Host temperature

Rewwan, 2024-04-03:
How can I add a temperature display for the temperature sensor built into t113-s3?

consp, 2024-04-03:
You'd need to add lm-sensors to the build system but if I'm not mistaken the sensor is not included in the flashforge kernel (which is still in use) and might also need the kernel module for it (which also needs to be compiled). @xblax do you know if it needs another module?

Afterwards you could add it to klipper.

xblax, 2024-04-03:
lm-sensors in Buildroot does not help, I already tried that - no sensor is found. Afaik I also looked in to the Kernel source to see if a missing module could be compiled, but did not find an easy solution.

Afaik the CPU temp sensor would have to be configured in the device tree: https://www.kernel.org/doc/Documentation/devicetree/bindings/thermal/thermal.txt

I pretty much concluded that it's not worth the effort.

consp, 2024-04-03:
```
		ths@02009400 {
			compatible = "allwinner,sun8iw20p1-ths";
			reg = <0x00 0x2009400 0x00 0x400>;
			clocks = <0x02 0x54>;
			clock-names = "bus";
			resets = <0x02 0x23>;
			nvmem-cells = <0x63>;
			nvmem-cell-names = "calibration";
			#thermal-sensor-cells = <0x01>;
		};
```

It's commented in the devicetree so it would require a change to that. Pretty much a no-go I guess.

Rewwan, 2024-04-03:
> ```
> 		ths@02009400 {
> совместимость = "allwinner,sun8iw20p1-ths";
> reg = <0x00 0x2009400 0x00 0x400>;
> часы = <0x02 0x54>;
> часы-имена = "автобус";
> сбросы = <0x02 0x23>;
> nvmem-клетки = <0x63>;
> nvmem-cell-имена = "калибровка";
> #тепло-сенсор-клетки = <0x01>;
> };
> ```
> 
> Он присутствует в дереве устройств, возможно, он может работать. Однако термическая зона не определена.
> 
> ```
> 10603225 0xA1CAD9 Unix путь: /домой/linux/работа/hard_t113_s3/t113/T113/T113-Linux-Auto-v1.1/ядро/linux-5.4/водители/тепловой/тепловой.c
> 10604169 0xA1CE89 Unix путь: /домой/linux/работа/hard_t113_s3/t113/T113/T113-Linux-Auto-v1.1/ядро/linux-5.4/водители/тепловой/cpu_cooling.c
> ```
> 
> Компилируются в (не имеют конфигурации прямо сейчас), я ожидал бы также, что там будет драйвер sun8i_thermal, если бы он еще не был включен в ядро. Был добавлен jan-2023 afaik. (примечание: поддержка t113/d1 была добавлена тогда)

So there is hope?

consp, 2024-04-03:
> So there is hope?

Not really. sunxi_thermal is compiled into the kernel, the device tree entry is there yet it's not populated so my guess would be the configuration is wrong. It will probably send some messages very ealy in kernel boot but that's hard to find out as /tmp is not yet populated and there are no logs yet.

closed 2025-09-16 by xblax