xblax/flashforge_ad5m_klipper_mod #135: Incorrect name?

skylatian, 2024-04-06:
This isn't exactly the biggest issue, but it looks like the repo switches the "5" and "m" in the name. It should be "ad5m" not "adm5". Feel free to close this if it's not a priority to fix, thought I should mention it!

xblax, 2024-04-08:
Yes it's technically incorrect. But cannot be changed since that would break a lot of links. It should not really matter in the end.
I will do a grep over the repo and make sure it's correct otherwise.

skylatian, 2024-04-08:
Unfortunately, the name is messing with searchability on Google - the searching for "AD5M klipper" really only reveals @g992's research repo. To be fair, that does link here, but it's not immediately obvious. If you're referring to links to the repository breaking, Github actually handles redirects to renamed repos pretty cleanly - According to [this comment](https://github.com/orgs/community/discussions/22669#discussioncomment-3237667.), links to renamed repos will be redirected to the correct location unless a repo of the same name is made in place of the old one.

I renamed my fork of this repo to `flashforge_ad5m_klipper_mod` to check, and it works well:
Old link: https://github.com/skylatian/flashforge_adm5_klipper_mod
Correctly redirects to: https://github.com/skylatian/flashforge_ad5m_klipper_mod

consp, 2024-04-08:
> I renamed my fork of this repo to flashforge_ad5m_klipper_mod to check, and it works well:

You can dual name if needed -> https://docs.github.com/en/repositories/creating-and-managing-repositories/renaming-a-repository

xblax, 2024-04-08:
Ok that's good to know. I will look into it and will then probably rename it.

Until some while ago this repo was not indexed by Google at all. I believe that is because links to this repo were shared before it was public but returned error 404 for everyone that wasn't a collaborator.

I tried registering it for Google search console, but that's not possible since the verification methods don't work with Github repo. It would only work for HTML pages that are generated from the docs.

consp, 2024-04-08:
As soon as it's mentioned anywhere (e.g. reddit) it will be followed and checked again, might take a while to show up though.

xblax, 2024-04-08:
Ok, renaming was quite unspectacular. Github just forwards everything with HTTP 302 Moved. Google should pick this up quickly.
The submodules must still be renamed and https://pypi.org/project/ff-adm5-audio/ project also adopts that common mistake.

I guess it was wrong in more places than it was right. Should be mostly consistent now.

consp, 2024-04-09:
> The submodules must still be renamed and https://pypi.org/project/ff-adm5-audio/ project also adopts that common mistake.

I added them both to avoid typosquatting and they are linked to my api keys. You can pick either one.

xblax, 2024-05-30:
All the renaming was completed quite some time ago. This can be closed.

closed 2024-05-30 by xblax