xblax/flashforge_ad5m_klipper_mod #141: Macro nomenclature

KaruroChori, 2024-04-12:
Orcaslicer by default in its basic profiles uses `PRINT_START EXTRUDER=* BED=*`, `PRINT_END` and `PAUSE` for a klipper profile.
The macro which have been implemented for the flashforge follow a similar yet incompatible nomenclature.

I would suggest to make it uniform with the de facto standard so that one can use the same profile with multiple printers.

xblax, 2024-04-12:
There is not really an agreed standard, I think. The configuration for this mod follows the suggested scheme from the Klipper documentation: https://www.klipper3d.org/Slicers.html?h=slicer#start_print-macros

I'm not sure it's a good idea to change it now, because that will mess with the existing G-Codes and Profiles that were already created for this mod.

KaruroChori, 2024-04-12:
What a mess. I assumed Orcaslicer was following Klipper's own guidelines in a klipper profile.
It would have been possible to just create an alias, but at this point it does not really make sense, it is just Orcaslicer which should avoid to reinvent the wheel in naming conventions.

closed 2024-04-12 by KaruroChori