xblax/flashforge_ad5m_klipper_mod #147: Doesnt with with latest firmware

insp1re1, 2024-04-16:
Doesnt work with latest 2.6.5 firmware.

xblax, 2024-04-16:
Yes that is known already. See discussion https://github.com/xblax/flashforge_ad5m_klipper_mod/discussions/144
We will enable support for the latest firmware in the next release.

insp1re1, 2024-04-17:
Could try leaving a note on the releases page, saves people wasting time.

KaruroChori, 2024-04-17:
From https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/master/docs/INSTALL.md, one of the first few lines:
> The mod installer currently requires that printers were updated to version 2.4.5 of the stock Flashforge firmware.

It is explicitly reported to work for 2.4.5, and the user is requested to have it installed.
If people cannot read the documentation, there is no way to prevent them from wasting time :/

closed 2024-04-17 by KaruroChori

insp1re1, 2024-04-17:
isnt mentioned in documentation for the install, what the install docs say is "The mod installer currently requires that printers were updated to version 2.4.5 of the stock Flashforge firmware", one would assume this version as a baseline.

pillock.