xblax/flashforge_ad5m_klipper_mod #150: Must home axis first: 105.000 105.000 220.000 [0.000]

jvarrone, 2024-04-19:
First time trying to print with Orca 2.0.0 after mod installed, I keep getting the error:  _"Must home axis first: 105.000 105.000 220.000 [0.000]"_ when I try to print any file on the ADM5 Pro. It happens immediately after the bed reaches its set temp, right before the nozzle begins to warm up.

I also tried the Klipper Lite mod with no success. Consoel output below

I searched the net and found several articles about BED_MESH_CLEAR and I tried to include it in the gcode but no success. It is likely I don't know where to put it.

Any ideas? I'm stuck.

Console Output:
17:55:04 File opened:NikonBatteryCap1.2_TPU_12m47s.gcode Size:226012
17:55:04 File selected
17:56:09 !! Must home axis first: 0.000 0.000 5.000 [0.000]
17:56:09 echo: "extruder" not hot enough
17:56:09 echo: Printer not homed
17:56:09 echo: "extruder" not hot enough
17:56:09 !! Must home axis first: 105.000 105.000 220.000 [0.000]
17:56:09 !! Must home axis first: 105.000 105.000 220.000 [0.000]
17:56:09 !! Must home axis first: 105.000 105.000 220.000 [0.000]
17:56:09 !! Must home axis first: 105.000 105.000 220.000 [0.000]

​

AJolly, 2024-04-19:
You need to set your orca machine gcode to either use the klipper macros or write your own. (For start and end)

AJolly, 2024-04-19:
https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/master/docs/SLICING.md

jvarrone, 2024-04-19:
Gcode flavor was set to Klipper.  Sorry for the NOOB question, but which specific config file is it that I need to modify to put the start/end code that's in the link?

consp, 2024-04-20:
> but which specific config file is it that I need to modify to put the start/end code that's in the link?

With most slicer variants it's under printer config -> machine gcode (I think it's on the top of the tab in orca)

Tiwatz, 2024-04-20:
See [#79](https://github.com/xblax/flashforge_ad5m_klipper_mod/discussions/79) for the Orca start and end g-gode

jvarrone, 2024-04-20:
THANK YOU for this! I was on the edge of giving up. 
What I don't understand is that OrcaSlicer works without this additional code on the stock Flashforge Firmware (2.4.5 and 2.4.6). Couldn't this FW Mod perform those commands if they're not present in the code files it is sent? It's my understanding that they're needed for the FF printer and this FW Mod  _is for_ that printer.

jvarrone, 2024-04-22:
Please adjust the FW mod to compensate for this so that we don't have to add code to our slicers.

KaruroChori, 2024-04-22:
The firmware of the printer cannot compensate for the wrong slicer profile being used.
The best we can do is to share profiles for the modded version to be used in place of having the user manually changing them.

jvarrone, 2024-04-22:
The OEM (2.45 AND 2.46) Flashforge FW works with my OrcaSlicer, so I don't understand why the MOD doesn't, since you are modding the Klipper FW for the Flashforge printer. Please help me understand.

KaruroChori, 2024-04-22:
@jvarrone 
Because a choice was made, to diverge from the original flashforge implementation if needed.
This is to more closely follow the general guidelines offered by klipper in their documentation, allowing this printer to behave more like any other klipper printer. This affected the start and stop routines for jobs as well, which are no longer just a bunch of GCODE commands, but custom klipper commands.

Flashforge designed a custom proprietary user interface built on top of klipper, and they also have their customized slicer, so there was no reason for them to care about further user needs, customization and interoperability with different software. However the objectives of this projects do not perfectly match with those of flashforge; as such the outcome is from time to time different.

TLDR: This is to ensure a wider flexibility & compatibility of this printer with the rest of the klipper ecosystem. All this **is documented**, and it is the user responsibility, even more so when playing with a beta piece of software to have enough care to read it.

jvarrone, 2024-04-22:
Thanks for that. I have been reading the docs and learning as I go along. Is there a place or a file where the needed START_/END_ PRINT commands can be placed -once- instead of having to edit every profile I create? If so, I have not been able to find the .cfg files referenced in the various posts I've read or the docs. Again, thanks for this.

KaruroChori, 2024-04-22:
@jvarrone just to clarify this further: you might be under the wrong impression that the original flashforge firmware does not require any addidional start/finish gcode. This is no true, you are using their profile for their printer, specifically because they shipped the required commands in there.
What we did was move them back to the printer as per general guideline when configuring a klipper printer, and change the start/finish code in the slicer profile to match.

So the configuration for a new job has always been split between printer and slicer. We just changed the balance so to speak :).

Changes in the section hosted on the printer are part of the `[gcode_macro START_PRINT]` and `[gcode_macro END_PRINT]`. I would advice against touching them unless you fully understand what you are doing.
Changes on the slicer side depend on the slice and must be manually set up for each profile.
[image]
Please ignore the specifics of that configuration as it was done for a different printer which was not even based on klipper.

jvarrone, 2024-04-22:
Now I understand 💡

Thank you!

 

image.png (view on web) <https://github.com/xblax/flashforge_ad5m_klipper_mod/assets/9377545/452c7a23-9a25-4ac9-b6dd-b3cde6435bd7>

Tiwatz, 2024-04-22:
> Is there a place or a file where the needed START_/END_ PRINT commands can be placed -once- instead of having to edit every profile I create?

You set it in the printer settings in Orca and not in the filament profile, as previously mentioned in this thread by @KaruroChori. So you only have to do it once and not for every filament profile.

jvarrone, 2024-04-22:
I followed that thread, and it said to add it to the macros.cfg file, which I do not have. Since this is critical to getting Orca and the mod to work, shouldn't it be absolutely clear in the docs "where" to make the edit? I am still lost.

Tiwatz, 2024-04-22:
You don't add anything to the macros.cfg file. In Orca slicer you just add the lines shown in the picture in:
https://github.com/xblax/flashforge_ad5m_klipper_mod/discussions/79#discussioncomment-8826586

jvarrone, 2024-04-22:
- So that goes in EACH Nozzle preset? I don't see one place for the entire machine, right?
- How about Filament Swap? I can't find that button on the panel.

Tiwatz, 2024-04-22:
> * So that goes in EACH Nozzle preset? I don't see one place for the entire machine, right?

Red arrow is for entire machine. 

[image]

jvarrone, 2024-04-22:
We are back to my original question. That only changes it for that nozzle (and saves it as a new preset). If I edit another nozzle, i.e. the .25 nozzle, my edit of the other nozzle is not there. That seems to be correct because the edit is for the nozzle. Not the entire machine. I can send screenshots if you like.

KaruroChori, 2024-04-22:
I see what you mean.
Sadly the tooling around 3d printers is far from perfect.
There is no good way to handle the inventory of filaments, flow calibration in the machine, the generation of fully portable gcode, and the decoupling of the printer configuration from the nozzle. 

So yes, that is just a limitation of the software we have available at this point in time.

jvarrone, 2024-04-22:
Such a file might look like this internally and be applied to all gcode being sent to the printer.

 

[gcode Start]

START_PRINT

 

[gcode End]

END_PRINT

KaruroChori, 2024-04-22:
I mean, there is: in theory start_print, end_print etc do that.
However, that is not some standard gcode, just some klipper macro extensions whose nomenclature is actually arbitrary, and the user is forced to recall those macro in the slicer profiles to have them running.

Default gcode has really no way do understand what starting or ending a job means, as there is no primitive to represent the very concept of a job to begin with.

While I agree this is bad design-wise, it is just the legacy of using a 30 years old language with some incremental improvements patched on top. Definitely something this repository cannot solve :D.

jvarrone, 2024-04-22:
Yup, I understand. The change I suggested needs to be on the Orca side. 

 

I’m putting this FW mod through its paces. So far, after this issue, I still can’t find a way to change filaments from the front panel, or anywhere for that matter.

KaruroChori, 2024-04-22:
Mid print? If so please open a new issue to track that one specifically.