xblax/flashforge_ad5m_klipper_mod #154: Firmware crashing consistently when info button hit.

jvarrone, 2024-04-20:
When I hit the little "i" in the lower left hand of the screen, the firmware crashes consistently. Please see the screenshots.

[image]
[image]

KaruroChori, 2024-04-20:
Is it not what it is supposed to be? I don't use klipperscreen but isn't it the halt function?

Tiwatz, 2024-04-20:
That is not an I, it is a exclamation mark. And the "info button" is the emergency stop

KaruroChori, 2024-04-20:
As a I thought, so there is no bug there.

jvarrone, 2024-04-20:
Sorry, the message it threw about a webhook misled me into believing it was an issue. Perhaps consider making the resulting message more indicative of what actually took place.

 

Thanks for a great mod.

KaruroChori, 2024-04-20:
This repository does not mod the behaviour of other programs unless it is needed to fit within the limited hardware of this printer.
You should direct this feedback [here](https://github.com/KlipperScreen/KlipperScreen) as they are the maintainers of klipperscreen.

closed 2024-04-21 by KaruroChori