xblax/flashforge_ad5m_klipper_mod #156: Move stepper z position_endstop: out of printer.base.cfg

AJolly, 2024-04-22:
I suggest moving position_endstop: 220 out of printer.base.cfg to printer.cfg  
That way you can save the endstop if you change it by z-adjusting.

For mine it was closer to 221.7

xblax, 2024-04-24:
I don't think it should be changed. The purpose of the endstop is only to have a consistent reference position and everything else is determined by the bed mash. If we change the endstop so that the nozzle is directly at the bed at position z=0 even without bed mesh, this can lead to issues when you change the print surface.

For printers like the AD5M, we only need to create a bed mesh and adjust the probe offset (not the endstop). This is different from printers where the probe is also the endstop.

xblax, 2024-05-30:
I'm going to close this, due to the reason explained in the last comment.

Since the last release we also replace the Z_OFFSET_APPLY_ENDSTOP command with Z_OFFSET_APPLY_PROBE to avoid that users unexpectedly run in to this issue:
https://github.com/xblax/flashforge_ad5m_klipper_mod/commit/b83cb9bc4c206984c5304a4c95506426149ad431

closed 2024-05-30 by xblax