xblax/flashforge_ad5m_klipper_mod #164: Message: Print is not paused, resume aborted

Tiwatz, 2024-04-26:
My print paused at 97% today and it was impossible to start again. As soon as I hit the "Resume print" button in Mainsail the Console message was "Print is not paused, resume aborted". I could not find that message in any of the config files. 
[image]

[klipper.log](https://github.com/xblax/flashforge_ad5m_klipper_mod/files/15135606/klipper.log)
[moonraker.log](https://github.com/xblax/flashforge_ad5m_klipper_mod/files/15135607/moonraker.log)

xblax, 2024-05-05:
I think I've seen this issue mentioned somewhere else before, that the printer went into Pause for unknown reason. For the first time I did now also encounter this issue with an overnight print.

And I think it's caused by this (from klipper log)
```
virtual_sdcard read
Traceback (most recent call last):
  File "/root/printer_software/klipper/klippy/extras/virtual_sdcard.py", line 237, in work_handler
    data = self.current_file.read(8192)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen codecs>", line 322, in decode
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x9c in position 4315: invalid start byte
Exiting SD card print (position 5865465)
```

Your log shows a similiar error. Thanks for attaching it:
```
virtual_sdcard read
Traceback (most recent call last):
  File "/root/printer_software/klipper/klippy/extras/virtual_sdcard.py", line 237, in work_handler
    data = self.current_file.read(8192)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen codecs>", line 322, in decode
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x9f in position 136: invalid start byte
Exiting SD card print (position 10854385)
```

We have to check if it's a known issue on klipper side or think what else could cause this.

xblax, 2024-05-05:
I attempted to print the same file again and it fails at exactly the same position.
```
virtual_sdcard read
Traceback (most recent call last):
  File "/root/printer_software/klipper/klippy/extras/virtual_sdcard.py", line 237, in work_handler
    data = self.current_file.read(8192)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen codecs>", line 322, in decode
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x9c in position 4315: invalid start byte
Exiting SD card print (position 5865465)
```
I had it uploaded directly from Prusa slicer. I have to investigate if it's actually corrupted.

consp, 2024-05-05:
`0x9c` is œ. but I doubt it's ascii/latin1 encoded.

Rand it though xxd and then looked at the results:

```
Line 678351: 00a59ce0: 4c49 4d49 5420 4143 4345 4c3d 3230 3030  LIMIT ACCEL=2000
Line 678352: 00a59cf0: 0a47 3120 4639 3130 392e 3832 370a 4731  .G1 F9109.827.G1
Line 678414: 00a5a0d0: 2129 7845 583d 2f55 e89c 25a5 4865 5026  !)xEX=/U..%.HeP&
Line 678424: 00a5a170: e89c 25a5 4865 7026 0000 19a4 0020 1901  ..%.Hep&..... ..
Line 678433: 00a5a200: 2129 7845 583d 2f55 e89c 25a5 4865 8026  !)xEX=/U..%.He.&
Line 678438: 00a5a250: 4c1d 19cc c482 e0be 6000 c480 0190 0021  L.......`......!
Line 678443: 00a5a2a0: e89c 25a5 4865 b026 0000 1da4 0020 1901  ..%.He.&..... ..
Line 678452: 00a5a330: 2129 7845 583d 2f55 e89c 25a5 4865 c026  !)xEX=/U..%.He.&
Line 678557: 00a5a9c0: 332e 3239 3920 4a37 2e36 3038 2045 2e30  3.299 J7.608 E.0
Line 678813: 00a5b9c0: 312e 3732 3120 5934 2e37 3720 452e 3230  1.721 Y4.77 E.20
Line 678957: 00a5c2c0: 3782 df81 9c40 6a16 6000 c480 0190 0021  7....@j.`......!
Line 678958: 00a5c2d0: 6e02 89c0 1310 0000 0004 fe3f 887f 0e00  n..........?....
Line 678962: 00a5c310: e89c 25a5 4865 e026 0000 20a4 0020 1901  ..%.He.&.. .. ..
Line 678967: 00a5c360: 6000 c480 0190 0021 6f02 89c0 1310 0000  `......!o.......
Line 678971: 00a5c3a0: 2129 7845 583d 2f55 e89c 25a5 4865 f026  !)xEX=/U..%.He.&
Line 678977: 00a5c400: 7002 89c0 1310 0000 0004 fe3f 887f 0e00  p..........?....
Line 678981: 00a5c440: e89c 25a5 4865 0027 0000 22a4 0020 1901  ..%.He.'..".. ..
Line 679069: 00a5c9c0: 3832 2059 3635 2e37 3039 2045 2e31 3937  82 Y65.709 E.197
```

xblax, 2024-05-05:
Yes my GCode file is corrupted as well. I don't have the original to compare since I directly uploaded from Prusa Slicer.
The one attached here is from Orca Slicer, so at least it does not seem to be a specific issue.

@Tiwatz did you also directly upload the file from the slicer?

Tiwatz, 2024-05-05:
> @Tiwatz did you also directly upload the file from the slicer?

Yes I did. No more issues since this one time though.

consp, 2024-05-05:
Looks to me like a string offset error (oob) and the extra data is random memory data. Surprised it didn't crash orca if that is the case.

thhdragon, 2024-05-17:
could this be related to what appears as a max queue overflow error on stock fw and pauses the print? in the 3dp discord we were able to the issue to the gcode having too many/too fine arcs from the arc fitting setting thats enabled default in orcaslicer. i see that print has a lot of circles and the gcode is full of g3 moves. try reslicing with arc fitting disabled. klipper doesnt use arcs like marlin does. i think the 0.12 default resolution in orcaslicer is fine for our boards. if not related i still wanted to make you aware of the gcode arcs causing max queue overflow on stock in case that helped troubleshoot any failed prints on this build that might not have to do with the firmware