xblax/flashforge_ad5m_klipper_mod #165: Warning when using Mainsail witht his mod

jvarrone, 2024-04-27:
Getting this warning msg on the Mainsail Device tab in OrcaSlicer:

_Dependency: Klipper
The current Klipper version does not support all features of Mainsail. Update Klipper to at least v0.11.0-257._

consp, 2024-04-27:
While it might be annoying, there is little which can be done at this time: the klipper mcu code is provided by flashforge and it's currently impossible to update it. The host side already is at the latest compatible version (with some backports from klipper v0.12 in the next release)

Maybe it's possible to suppress the message for now.

xblax, 2024-05-06:
There is not much we can do about this. It's not worth patching this in the Mainsail code. Afaik beeing a bit behind with the Klipper version doesn't cause any issues and the message does only show up once if acknowledged and checked to not show again.

I'm going to Close it as "Won't Fix".

closed 2024-05-06 by xblax