xblax/flashforge_ad5m_klipper_mod #171: Auto turn off temps after a while

ugryy, 2024-05-04:
Hi there, firstly thanks for the mod! I enjoy klipper ^^
I am printing ASA and ABS these days. Before and between the prints I want to keep inside warm (I encountered some warping in some parts -2x3x1cm 25x quantity parts). 
I set the bed temp to 90 on mainsail, after like 10 minutes bed switches off. How can I adjust the time of auto-off?

Tiwatz, 2024-05-04:
Not sure exactly where right now but look for a `[idle_timeout]` section in one of the `.cfg` files.

ugryy, 2024-05-04:
I already checked like u said actually, there is quite a bit idle_timeout codes but I am not sure which one to edit. Can't see a value in minutes or hour.

found this one but I don't think this is it
[delayed_gcode AUTO_SHUTDOWN_CHECK]
gcode:
  {% if printer.idle_timeout.state == "Idle" or printer.idle_timeout.state == "Ready" %}
    {% if printer.extruder.temperature < 50.0 and printer.heater_bed.temperature < 50.0 %}
        {% if printer.extruder.target == 0.0 and printer.heater_bed.target == 0.0 %}
            UPDATE_DELAYED_GCODE ID=AUTO_SHUTDOWN_CHECK DURATION=0
            _SHUTDOWN_STAGE
        {% else %}
            UPDATE_DELAYED_GCODE ID=AUTO_SHUTDOWN_CHECK DURATION=2
        {% endif %}
    {% else %}
        {% if printer.idle_timeout.state == "Printing" %}
            UPDATE_DELAYED_GCODE ID=AUTO_SHUTDOWN_CHECK DURATION=0
        {% else %}
            {% if printer.extruder.target == 0.0 and printer.heater_bed.target == 0.0 %}
                UPDATE_DELAYED_GCODE ID=AUTO_SHUTDOWN_CHECK DURATION=2
            {% else %}
                UPDATE_DELAYED_GCODE ID=AUTO_SHUTDOWN_CHECK DURATION=0
            {% endif %}
        {% endif %}
    {% endif %}
  {% endif %}

Tiwatz, 2024-05-04:
`printer.base.cfg`

```
[idle_timeout]
timeout:600
```

The value is in seconds. I usually use 3600, that is 1 hour before the printer timeouts.

closed 2024-05-05 by ugryy