xblax/flashforge_ad5m_klipper_mod #174: Corrupted printer.cfg

KaruroChori, 2024-05-06:
Some operations performed via the guppyscreen interface resulted in an empty printer.cfg being saved.
I cannot replicate the exact steps which led to this.

Additional investigation is needed.

KaruroChori, 2024-05-06:
It is possible it relates with https://github.com/xblax/flashforge_ad5m_klipper_mod/issues/93
Or that it is a separate issue.

consp, 2024-05-06:
> It is possible it relates with #93 Or that it is a separate issue.

Might be the same, might be different. Never had the corruption, do you have the guppyscreen log file (in printer_data/logs)?

KaruroChori, 2024-05-06:
Sadly no, I reinstalled the mod to work with https://github.com/xblax/flashforge_ad5m_klipper_mod/discussions/8#discussioncomment-9330016

consp, 2024-05-06:
I'm also pretty sure guppy does not edit printer.cfg but mainsail/klipper does. So I guess it's a klipper/mainsail issue more than any ui issue.

KaruroChori, 2024-05-06:
Ok I will remove that tag, and close it as duplicate of #93 as they are virtually indistinguishable.

closed 2024-05-06 by KaruroChori