xblax/flashforge_ad5m_klipper_mod #179: guppy screen wifi

insp1re1, 2024-05-09:
connecting to a wifi ap via guppy screen connection, while connects to the ap does not provide ip via dhcp like ethernet does.

consp, 2024-05-12:
Can you provide the output of `/etc/network/interfaces`? I think the wlan0 interface is lacking the auto-dhcp.

consp, 2024-05-29:
Any more info on this or is it fixed?

insp1re1, 2024-05-29:
Not fixed, also max extruder temp on guppy screen is 240, no good for
higher temp materials

Ptifiela, 2024-06-07:
Same problem here.

> ok, then something is obviously wrong. Since you can connect via ethernet, are you willing to check the following for me, it would help a lot?
> 
>     * connect ethernet
> 
>     * wait till it gets an ip
> 
>     * connect wifi
> 
>     * check if it gets an IP or not

You need to forget the wifi ap you are connecting to, but I'm able to get an ip after following your steps.

consp, 2024-06-10:
My guess now is WPA is acting up in (un)cooperation with the dhcp client after connecting without a "new" entry.

xblax, 2025-09-16:
Closing this. There is currently no upstream development for Guppyscreen, so it's unlikely that we can improve something if there are some glitches (also not really confirmed).