xblax/flashforge_ad5m_klipper_mod #183: Guppyscreen: Dismissing a prompt in fluidd/mainsail will leave it on screen and pressing a button will segfailt guppyscreen.

consp, 2024-05-13:
I get it. I also couldn't find anything definitive about the action:prompt messages. Thank you for your efforts. I'm just trying to help with what I can.

I repeated the test with two buttons. Again, it works if I dismiss the dialog from guppyscreen. When I dismiss it from Fluidd, it does not disappear from the screen. When I then tap the OK button on-screen, the button text changes and guppyscreen hangs.
[image]

Here's the macro I used:
```
[gcode_macro TEST_PROMPT]
gcode:
    RESPOND TYPE=command MSG="action:prompt_begin Test Prompt"
    RESPOND TYPE=command MSG="action:prompt_text This is a test prompt."
    RESPOND TYPE=command MSG="action:prompt_footer_button Abort|RESPOND TYPE=command MSG=action:prompt_end"
    RESPOND TYPE=command MSG="action:prompt_footer_button OK|RESPOND TYPE=command MSG=action:prompt_end"
    RESPOND TYPE=command MSG="action:prompt_show"
```

Here's the guppyscreen.log:
```
[2024-05-10 04:18:53.316] [guppyscreen] [debug] done init
[2024-05-10 04:19:00.206] [guppyscreen] [debug] data found
[2024-05-10 04:19:00.206] [guppyscreen] [debug] data: // action:prompt_begin Test Prompt
[2024-05-10 04:19:00.206] [guppyscreen] [debug] action: prompt_begin Test Prompt
[2024-05-10 04:19:00.206] [guppyscreen] [debug] PROMPT_BEGIN: Test Prompt
[2024-05-10 04:19:00.694] [guppyscreen] [debug] data found
[2024-05-10 04:19:00.694] [guppyscreen] [debug] data: // action:prompt_text This is a test prompt.
[2024-05-10 04:19:00.694] [guppyscreen] [debug] action: prompt_text This is a test prompt.
[2024-05-10 04:19:00.694] [guppyscreen] [debug] PROMPT_TEXT: This is a test prompt.
[2024-05-10 04:19:01.405] [guppyscreen] [debug] data found
[2024-05-10 04:19:01.405] [guppyscreen] [debug] data: // action:prompt_footer_button Abort|RESPOND TYPE=command MSG=action:prompt_end
[2024-05-10 04:19:01.405] [guppyscreen] [debug] action: prompt_footer_button Abort|RESPOND TYPE=command MSG=action:prompt_end
[2024-05-10 04:19:01.405] [guppyscreen] [debug] 26 -1
[2024-05-10 04:19:01.434] [guppyscreen] [debug] PROMPT_FOOTER_BUTTON: Abort CMD: RESPOND TYPE=command MSG=action:prompt_end, type none
[2024-05-10 04:19:02.157] [guppyscreen] [debug] data found
[2024-05-10 04:19:02.157] [guppyscreen] [debug] data: // action:prompt_footer_button OK|RESPOND TYPE=command MSG=action:prompt_end
[2024-05-10 04:19:02.157] [guppyscreen] [debug] action: prompt_footer_button OK|RESPOND TYPE=command MSG=action:prompt_end
[2024-05-10 04:19:02.158] [guppyscreen] [debug] 23 -1
[2024-05-10 04:19:02.158] [guppyscreen] [debug] PROMPT_FOOTER_BUTTON: OK CMD: RESPOND TYPE=command MSG=action:prompt_end, type none
[2024-05-10 04:19:02.652] [guppyscreen] [debug] data found
[2024-05-10 04:19:02.652] [guppyscreen] [debug] data: // action:prompt_show
[2024-05-10 04:19:02.653] [guppyscreen] [debug] action: prompt_show
[2024-05-10 04:19:02.653] [guppyscreen] [debug] PROMPT_SHOW
[2024-05-10 04:19:02.653] [guppyscreen] [debug] callback
[2024-05-10 04:19:08.380] [guppyscreen] [debug] Button 1 pressed
[2024-05-10 04:19:08.380] [guppyscreen] [debug] Command issued: RESPOND TYPE=command MSG=action:prompt_end
[2024-05-10 04:19:08.380] [guppyscreen] [debug] send_jsonrpc: {"id":98,"jsonrpc":"2.0","method":"printer.gcode.script","params":{"script":"RESPOND TYPE=command MSG=action:prompt_end"}}
[2024-05-10 04:19:08.882] [guppyscreen] [debug] data found
[2024-05-10 04:19:08.882] [guppyscreen] [debug] data: // action:prompt_end
[2024-05-10 04:19:08.882] [guppyscreen] [debug] action: prompt_end
[2024-05-10 04:19:08.882] [guppyscreen] [debug] PROMPT_END
[2024-05-10 04:19:16.310] [guppyscreen] [debug] data found
[2024-05-10 04:19:16.310] [guppyscreen] [debug] data: // action:prompt_begin Test Prompt
[2024-05-10 04:19:16.310] [guppyscreen] [debug] action: prompt_begin Test Prompt
[2024-05-10 04:19:16.310] [guppyscreen] [debug] PROMPT_BEGIN: Test Prompt
[2024-05-10 04:19:16.810] [guppyscreen] [debug] data found
[2024-05-10 04:19:16.810] [guppyscreen] [debug] data: // action:prompt_text This is a test prompt.
[2024-05-10 04:19:16.810] [guppyscreen] [debug] action: prompt_text This is a test prompt.
[2024-05-10 04:19:16.810] [guppyscreen] [debug] PROMPT_TEXT: This is a test prompt.
[2024-05-10 04:19:17.524] [guppyscreen] [debug] data found
[2024-05-10 04:19:17.524] [guppyscreen] [debug] data: // action:prompt_footer_button Abort|RESPOND TYPE=command MSG=action:prompt_end
[2024-05-10 04:19:17.524] [guppyscreen] [debug] action: prompt_footer_button Abort|RESPOND TYPE=command MSG=action:prompt_end
[2024-05-10 04:19:17.524] [guppyscreen] [debug] 26 -1
[2024-05-10 04:19:17.524] [guppyscreen] [debug] PROMPT_FOOTER_BUTTON: Abort CMD: RESPOND TYPE=command MSG=action:prompt_end, type none
[2024-05-10 04:19:18.240] [guppyscreen] [debug] data found
[2024-05-10 04:19:18.240] [guppyscreen] [debug] data: // action:prompt_footer_button OK|RESPOND TYPE=command MSG=action:prompt_end
[2024-05-10 04:19:18.240] [guppyscreen] [debug] action: prompt_footer_button OK|RESPOND TYPE=command MSG=action:prompt_end
[2024-05-10 04:19:18.240] [guppyscreen] [debug] 23 -1
[2024-05-10 04:19:18.240] [guppyscreen] [debug] PROMPT_FOOTER_BUTTON: OK CMD: RESPOND TYPE=command MSG=action:prompt_end, type none
[2024-05-10 04:19:18.732] [guppyscreen] [debug] data found
[2024-05-10 04:19:18.732] [guppyscreen] [debug] data: // action:prompt_show
[2024-05-10 04:19:18.732] [guppyscreen] [debug] action: prompt_show
[2024-05-10 04:19:18.733] [guppyscreen] [debug] PROMPT_SHOW
[2024-05-10 04:19:18.733] [guppyscreen] [debug] callback
[2024-05-10 04:19:19.249] [guppyscreen] [debug] data found
[2024-05-10 04:19:19.250] [guppyscreen] [debug] data: // action:prompt_end
[2024-05-10 04:19:19.250] [guppyscreen] [debug] action: prompt_end
[2024-05-10 04:19:19.250] [guppyscreen] [debug] PROMPT_END
[2024-05-10 04:19:28.038] [guppyscreen] [debug] Button 1 pressed
```

_Originally posted by @b-fitzpatrick in https://github.com/xblax/flashforge_ad5m_klipper_mod/discussions/134#discussioncomment-9409666_

consp, 2024-05-13:
Solution would be to properly handle prompt_end and avoid the double free. (or only use c++)

consp, 2024-05-29:
Will be fixed in next release (or if you download the guppyscreen file in the attached discussion and replace it)

xblax, 2025-09-16:
Guppy Screen in Klipper Mod was updated to latest master release, so this *should* be fixed, but didn't test it.

I'm closing the GuppyScreen issues, since it's unlikely they will be worked at by someone and upstream development has also stalled. I added some information about Guppy Screen "broken" state to [Known Issues](https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/master/docs/LINUX.md), so that it's clear it can be used as-is but will not be worked on at the moment.

closed 2025-09-16 by xblax