xblax/flashforge_ad5m_klipper_mod #186: Adding klipper KAMP

3DEngelen, 2024-05-18:
Would it be possible to add KAMP bed levelling?

KaruroChori, 2024-05-18:
Adding support for it would be possible:
- as long as it is and stays compatible with klipper 0.11
- if there is no assumption over the base system to be some debian derivative.

To be totally honest I do not see the value in it and I am not sure it should be part of a "neutral" distribution.
Aside from the basic fact that calibrating the bed for this printer has been pretty much a "do it once" thing for me.
At the very worst you might want few profiles to account for the different bed temperatures, and recall them via GCODE.

Have you tried to install it yourself and verified it works?

3DEngelen, 2024-05-18:
I was able to get KAMP installed and operational.

Just followed the install method and needed to adapt the print_start macro.

If helpfull i can add my configuration of it.

Avpman2, 2024-05-18:
yes, please on the configuration info.

3DEngelen, 2024-05-18:
So install as described here : [https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging](url)
And change the configs as exampled in the attachment. 
[KAMP_ADM5.zip](https://github.com/xblax/flashforge_ad5m_klipper_mod/files/15366199/KAMP_ADM5.zip)

The printer start G-code is:

M117 
START_PRINT BED_TEMP={first_layer_bed_temperature[0]} EXTRUDER_TEMP={first_layer_temperature[0]} FORCE_LEVELING=true

Also make sure to enable Label objects in the slicer (in orca it is in the Others tab of the print settings)

Avpman2, 2024-05-18:
Thanks! Looking forward to getting it set up.

consp, 2024-05-19:
Bit late to the party but it looks macro only, should be no issue I guess as it would be macro only. Would be quite easy to add (or for someone to add as a pull request) since it's only a change to `printer_config/*` files.

3DEngelen, 2024-05-19:
I did some more testing, and it seems that the location off the purge line is not correct when using KAMP.
My guess is that is due to the fact that the adventurer 5M uses the center of the bed as 0...

consp, 2024-05-19:
My guess would be it does not allow negative numbers since if it assumes a zero/zero corner origin and does some >0 filtering somewhere. Bit shame as it should not matter that much if you do the calculations correctly with respect to a normal cartesian coordinate system. Maybe ask the author about it.

thhdragon, 2024-05-21:
i think its these 2 lines right here.
`{% set purge_x_origin = ([purge_x_min - purge_margin, 0] | max) %}                                  # Add margin to x min, compare to 0, and choose the larger
    {% set purge_y_origin = ([purge_y_min - purge_margin, 0] | max) %}                                  # Add margin to y min, compare to 0, and choose the larger`
    
havent had a chance to test it but this would change the purge origin to 0 anytime its got negative coordinates which for us means the left and front of the bed. my guess is that it could work fine if you change 0 to something like -110

tr0tt3rs, 2024-05-21:
I feel like it’s more worthwhile to modify the AD5M config and set it up like most normal printers (with the 0,0 being in a corner rather than the center). Is there any good reason for setting 0,0 in the middle?

KaruroChori, 2024-05-21:
What do you mean like normal printers? All my printers are set with 0,0 in the middle. For deltas it is the only reasonable way to set the origin, and for the other types there is no reason not to :/.

thhdragon, 2024-05-21:
probing 0,0 in the center is a delta thing i think. ive never used a bedslinger or a corexy that hasnt had 0,0 at the bottom left corner. square beds are usually bottom left corner

tr0tt3rs, 2024-05-21:
Interesting! I’ve only ever had voron printers before this and I always set up 0,0 at the corner of the build plate. Never knew about the differences for delta printers.

thhdragon, 2024-05-21:
https://www.klipper3d.org/Bed_Mesh.html#basic-configuration
i think its round beds not deltas actually

consp, 2024-05-21:
> my guess is that it could work fine if you change 0 to something like -110

Looks like this is an out-of-bounds filter, iirc you can get those bounds somewhere in klipper and maybe use those instead of 0.

Looks like some combination of https://www.klipper3d.org/Status_Reference.html?h=axis_maximum#toolhead and https://www.klipper3d.org/Status_Reference.html?h=axis_maximum#gcode_move might get the min/max values needed.

xblax, 2024-05-26:
> At the very worst you might want few profiles to account for the different bed temperatures, and recall them via GCODE.
> @xblax also seems to share the same position reading [here](https://github.com/xblax/flashforge_ad5m_klipper_mod/discussions/91).

Yep, I think it's not really useful due to the reasons already mentioned in that discussion.

I was also thinking about changing the origin, but decided to not do it to stay somewhat compatible with the stock printer profile. 

If we think about pre-sliced G-Code supposed to run a different printers, it would also make sense to have the origin in the center of the bed. Anyway, it doesn't make much sense to change it now, as it would break all existing G-Codes and printer profiles for the AD5M.

i0n1zeD, 2024-06-06:
> So install as described here : https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging And change the configs as exampled in the attachment. [KAMP_ADM5.zip](https://github.com/xblax/flashforge_ad5m_klipper_mod/files/15366199/KAMP_ADM5.zip)
> 
> The printer start G-code is:
> 
> M117 START_PRINT BED_TEMP={first_layer_bed_temperature[0]} EXTRUDER_TEMP={first_layer_temperature[0]} FORCE_LEVELING=true
> 
> Also make sure to enable Label objects in the slicer (in orca it is in the Others tab of the print settings)
> 
> @consp edit: changed url so it redirects to the actual url instead of 'url'

I tried your settings but ran into a problem.
Just before BED_MESH_CALIBRATE starts, it does not execute LOAD_CELL_TARE because the macro is overridden.
And as a result, we have a not very good first layer!

thhdragon, 2024-06-18:
> I tried your settings but ran into a problem. Just before BED_MESH_CALIBRATE starts, it does not execute LOAD_CELL_TARE because the macro is overridden. And as a result, we have a not very good [...]

try these. not necessary to use force_leveling=true because i made it default true. just changes for the adaptive mesh not the purge line
[kamp_ad5m.zip](https://github.com/user-attachments/files/15880706/kamp_ad5m.zip)

fernplant, 2024-11-23:
> So install as described here : https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging And change the configs as exampled in the attachment. [...]

Thanks for sharing your config. Just to clarify for my sake, you have adaptive mesh working and it uses the proper nozzle wipe routine (to clean nozzle for proper readings)?

luckyPipewrench, 2025-01-11:
I've got KAMP adaptive mesh working with the wipe before. But adaptive purge does not work correctly, it starts the purge in the middle of the bed.

fluffyeggs, 2025-01-21:
> I've got KAMP adaptive mesh working with the wipe before. But adaptive purge does not work correctly, it starts the purge in the middle of the bed.

Line_Purge.cfg needs any reference of 0 as origin changed to -110.  With these changes it will print the purge line in front of the object if there is room. 

`{% set purge_x_center = ([((purge_x_max + purge_x_min) / 2) - (purge_amount / 2), -110] | max) %}`
`{% set purge_y_center = ([((purge_y_max + purge_y_min) / 2) - (purge_amount / 2), -110] | max) %}`

`{% set purge_x_origin = ([purge_x_min - purge_margin, -110] | max) %}`
`{% set purge_y_origin = ([purge_y_min - purge_margin, -110] | max) %}`

Also, on line 61 and 83.
`{% if purge_y_origin > -110 %}`

[image]