xblax/flashforge_ad5m_klipper_mod #189: Unable to parse .02.48 Error during print

sethpark777, 2024-05-23:
Not quite sure if this is unique to the mod or some other known issue. But when printing, I got this error:
`Error on 'G2 X11.057 Y4.504 I-.226 J-1.334 E.02.48 Y4.364': unable to parse .02.48`

Just figured i would drop in the error, in case it is something known and i can be of help in some way. 
I have attached the gcode which i was printing at the time. 

[Shrek+Buddha_PLA_54m25s.zip](https://github.com/xblax/flashforge_ad5m_klipper_mod/files/15416631/Shrek%2BBuddha_PLA_54m25s.zip)

KaruroChori, 2024-05-23:
Yeah, .02.48 is really not a number :D. Maybe a bug in the slicer?

sethpark777, 2024-05-23:
You know, somehow my brain totally ignored that leading decimal point and zero. 

I agree, looks to be a slicer bug. I'm running 2.0 orca, no post processors. So can go put one in over there!

closed 2024-05-23 by sethpark777