xblax/flashforge_ad5m_klipper_mod #191: Mainsail urls cannot be accessed directly.

AJolly, 2024-05-23:
I'm not sure why but mainsail urls from this project can't ever be accessed directly.
For example 
http://192.168.40.220:4000/console
or http://192.168.40.220:4000/heightmap

It gives a 404 not found message, unless I open the root url and then browse to it.
This means I can't make any bookmarks.

Another Klipper machine I set up using kamp works perfectly without needing to browse to the main url first.

consp, 2024-05-26:
I have not yet verified it but it might be the http server. Most debian machines use nginx which is configured differently than the http server in use.

xblax, 2024-05-26:
Yep, probably there are some URL rewrite rules in the recommended nginx config. We can have a look if it's possible to configure something similar in busybox httpd.