xblax/flashforge_ad5m_klipper_mod #193: Power button on pro fails to respond after several pushes

consp, 2024-05-29:
Cannot replicate it reliably but something like the following happens:

* press button
* popup appears
* cancel
* print more
* press button
* popup appears
* cancel again (still ... need ... to ... print ... more)
* press button
* nothing happens

It does not look like a prompt issue, either klipper state or button state is broken at that point. Rebooting fixes the problem.

If anyone else has this issue, please leave a message with the circumstances.

Avpman2, 2024-06-05:
Similar issue with the Extrude/Retract buttons. Sometimes, it works, but frequently not. Especially if you have need to extrude more than once in a row.

consp, 2024-06-05:
> full Klipper is sluggish in responding to button presses

Unfortunately that's due to klipperscreen method of allocating everything in dictionaries and being slow in general ... and the cpu is not helping much unfortunately. Unless someone finds a method of speeding up klipperscreen by magical means I'm pretty sure that's the max it can do.

The flashforge GUI is written in c++ (with Qt4), like guppyscreen (with lvgl which is even leaner) and I guess that's the way to go for a decent "responsive" experience.

My advice would be to be a bit more patient despite that being sort-of a crude way of saying "wait a bit".

Avpman2, 2024-06-05:
Thanks! I'll try Guppyscreen again once it's out of Alpha.