xblax/flashforge_ad5m_klipper_mod #194: Can't Install On Recent Firmware

WatWall, 2024-05-29:
I was able to install the klipper mod before v2.6.5 but I can't install it at the moment. I tried installing the guppyscreen and the lite version but printer just boots as usual. It doesn't ask or do something about firmware installation.

WatWall, 2024-05-29:
I successfully installed it after doing these 2 things, not sure which one fixed the issue.
First I Installed klippermod_uninstall then I installed guppyscreen using another usb stick

consp, 2024-05-30:
Interesting, though I doubt it's the guppyscreen version specific since the install scripts are the same.

DylanJMas, 2024-06-02:
Same thing for me,  just the USB, for some reason Flashforge printers is iffy on what USB drives you use, (older the better I think)

consp, 2024-06-05:
Most likely an exfat formatted USB drive, all new ones are. It must be FAT32 to work properly (and possibly also MBR but that's less likely). This is a constraint of the installed uboot which we cannot do anything about.

closed 2024-06-05 by consp