xblax/flashforge_ad5m_klipper_mod #195: Klipper has shutdown due to webhooks request

RyantHults, 2024-05-30:
Just installed v00.05-beta on my A5M, which was running 2.6.6. Everything seemed to install correctly, however immediately after everything starts up, Klipper shuts down.
The status page says: 
"Klipper has shutdown

Shutdown due to webhooks request. 
Once the underlying issue is corrected
use the FIRMWARE_RESET command to reset the firmware, reload the config, and restart the host software. 
Printer is shutdown
"
I've tried restarting the printer, restarting the host, and fully powering off the back on the A5M. All result in the same error.

RyantHults, 2024-05-30:
ok, turns out this is exactly what is expected, I just didn't realize how things were laid out in klipperscreen. I've never used klipperscreen before, and I was clicking the emergency stop button the main screen thinking it was an error indicator and clicking on it would take me to like a log page or something. Might be a good idea to update that icon to the one klipperscreen uses now to reduce confusion. 
[image]

closed 2024-05-30 by RyantHults

consp, 2024-05-30:
Ok, did the same thing first time. You are not alone 😂