xblax/flashforge_ad5m_klipper_mod #196: Terrible printing with OrcaSlicer and Klipper for AM5P

f00d4tehg0dz, 2024-05-31:
All filament has been dried in several dry boxes. However, every filament refuses to stick on the bed plate for an extended period, and filament clogs even after switching out nozzles—thin, stringy extrusion. 

Prints work fine when switching back to Flashforge AM5P stock firmware and OrcaSlicer Flashforge profile.

I have implemented START_PRINT and END_PRINT in appropriate locations and have done a mesh calibration profile

Any thoughts on what I'm doing wrong?

KaruroChori, 2024-05-31:
Please share the gcode used for the stock firmware and the one for the modified firmware.

Avpman2, 2024-05-31:
Which specific version of the FW mod are you using? There is a guppy screen, light and full. I had difficulty with the guppy screen version and when I switched back the the full implementation the issues went away. I think it might have been a config file issue, and not related to the build. Also, check your "bed Type" settings. If it's set for a smooth plate and you're using a textured PEI (and vice versa), there will be problems.

f00d4tehg0dz, 2024-06-01:
> Please share the gcode used for the stock firmware and the one for the modified firmware.

Let me know if I'm providing the wrong g-codes here.

**Flashforge Adventurer 5MP 0.4mm Default on OrcaSlicer**

```
Machine start G-code
M190 S[bed_temperature_initial_layer_single]
M104 S[nozzle_temperature_initial_layer]
G90
M83
G1 Z5 F6000
G1 E-1.5 F800
G1 X110 Y-110 F6000
G1 E2 F800
G1 Y-110 X55 Z0.25 F4800
G1 X-55 E8 F2400
G1 Y-109.6 F2400
G1 X55 E5 F2400
G1 Y-110 X55 Z0.45 F4800
G1 X-55 E8 F2400
G1 Y-109.6 F2400
G1 X55 E5 F2400
G92 E0

Machine end G-code
G1 E-3 F3600
G0 X50 Y50 F30000
M104 S0 ; turn off temperature

Before layer change G-code
;BEFORE_LAYER_CHANGE
;[layer_z]

Layer change G-code
;AFTER_LAYER_CHANGE
;[layer_z]

```

**Flashforge Adventurer 5MP 0.4mm Klipper on OrcaSlicer**

```
Machine start G-code
M190 S[bed_temperature_initial_layer_single]
M109 S[nozzle_temperature_initial_layer]
PRINT_START EXTRUDER=[nozzle_temperature_initial_layer] BED=[bed_temperature_initial_layer_single]

Machine end G-code
PRINT_END

Before layer change G-code
;BEFORE_LAYER_CHANGE
TIMELAPSE_TAKE_FRAME
;[layer_z]
G92 E0

Layer change G-code
;AFTER_LAYER_CHANGE
;[layer_z]

```

KaruroChori, 2024-06-01:
Please, check the correct syntax here: https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/master/docs/SLICING.md
Your args in the macro are not correct.

consp, 2024-06-01:
> There is a guppy screen, light and full. I had difficulty with the guppy screen version and when I switched back the the full implementation the issues went away.

That's intersting, the macros are the same.

f00d4tehg0dz, 2024-06-01:
> Please, check the correct syntax here: https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/master/docs/SLICING.md Your args in the macro are not correct. Because of that you are printing at the wrong temperature.

Thank you!

Avpman2, 2024-06-01:
P.S. It is important that you put the START_PRINT and EXTRUDER_TEMP, as indicated above, ON THE SAME line in your Gcode file. If they are on separate lines, your temperature will not be correct. Also, in the console output, you will see "EXTRUDER_TEMP" not found or something like that if they are on separate lines.

f00d4tehg0dz, 2024-06-02:
After following your advice I've been successfully printing all day. I'll close this out. Thanks!

closed 2024-06-02 by f00d4tehg0dz

Avpman2, 2024-06-02:
You are very welcome. I struggled with this myself until I finally figured it out!