xblax/flashforge_ad5m_klipper_mod #200: WIFI password that contains a space in it won't work

DylanJMas, 2024-06-02:
It took me forever to get my WIFI working but I figure it out. if your password is (Password) it would work correctly, but if the password is (Pass word) it would not do anything. But if you put it as ('pass word') it would work. I think what is happing is Linux only recognize the first word and not see the space as part of the password. 

maybe as the part of the code if the code sees a space it would automatically add "" as the output.

consp, 2024-06-05:
Could you elaborate on what method you used to enter the wifi password? Probably some escaping going wrong somewhere so it might be a bug which needs fixing.

DylanJMas, 2024-06-05:
I was using the printer interface. I tried to reproduce the problem, but it seems to work rn. When I first tried to set it up both versions of the interface wasn't working. 

I could l only could connect though Ethernet, but I couldn't ssh.  I look into the router, for some reason The router had two devices to the same IP. It was probably just bugging out because of that.

I reinstalled both versions, the guppy version worked fine but I still couldn't get the save button to do anything on the main version. The only way that I got the main one working is using the thermal and looking up how to Linux to accept a WiFi password that has space in it.

Sorry for wasting your time.

closed 2024-06-05 by DylanJMas

consp, 2024-06-05:
> Sorry for wasting your time.

No problem, since wifi is one of the buggiest things around (not just for the mod) with way too many different setups around the world it's good to have people figure out issues and if they are bugs, we need someone to figure out what is happening to fix it.