xblax/flashforge_ad5m_klipper_mod #201: verify_heater settings?

AJolly, 2024-06-02:
For verify_heater  / verify_extruder, whats the reason to change so far from Klipper defaults?

KaruroChori, 2024-06-02:
Can you better explain how these features are not aligned and what is the reference you are considering?

AJolly, 2024-06-05:
If we change away from default I'm usually interested in knowing why, and I couldn't find us having any discussion about it in this repo.
We use:  
[verify_heater extruder]
max_error: 290
check_gain_time:200
hysteresis: 10
heating_gain: 10

[verify_heater heater_bed]
max_error: 120
check_gain_time:200
hysteresis: 10
heating_gain: 10

https://github.com/Klipper3d/klipper/blob/master/docs/Config_Reference.md#verify_heater
Stock:[verify_heater heater_config_name]
#max_error: 120
#   The maximum "cumulative temperature error" before raising an
#   error. Smaller values result in stricter checking and larger
#   values allow for more time before an error is reported.
#   Specifically, the temperature is inspected once a second and if it
#   is close to the target temperature then an internal "error
#   counter" is reset; otherwise, if the temperature is below the
#   target range then the counter is increased by the amount the
#   reported temperature differs from that range. Should the counter
#   exceed this "max_error" then an error is raised. The default is
#   120.
#check_gain_time:
#   This controls heater verification during initial heating. Smaller
#   values result in stricter checking and larger values allow for
#   more time before an error is reported. Specifically, during
#   initial heating, as long as the heater increases in temperature
#   within this time frame (specified in seconds) then the internal
#   "error counter" is reset. The default is 20 seconds for extruders
#   and 60 seconds for heater_bed.
#hysteresis: 5
#   The maximum temperature difference (in Celsius) to a target
#   temperature that is considered in range of the target. This
#   controls the max_error range check. It is rare to customize this
#   value. The default is 5.
#heating_gain: 2
#   The minimum temperature (in Celsius) that the heater must increase
#   by during the check_gain_time check. It is rare to customize this
#   value. The default is 2.

xblax, 2024-06-05:
We did start with the stock config from Flashforge as a baseline. So the question should by why did they change it? Maybe they had false triggers with the Klipper default.

You can change it to the defaults and report here if you observe any change or issues.

xblax, 2025-09-16:
No change seems to needed, there are no reports that current config causes any issues.

closed 2025-09-16 by xblax