xblax/flashforge_ad5m_klipper_mod #203: MCU 'mcu' shutdown: Timer too close Adventurer 5M/5M pro

myxxx323, 2024-06-05:
Начал ловить зависания на Adventurer 5M pro и на Adventurer 5M. Установлен Adventurer 5M Klipper Mod v00.05-beta. Зависает в разном месте, разные детали. Слайсеры PrusaSlicer 2.7.4, OrcaSlicer 2.1.0 и Orca-Flashforge 1.0.2 . Печать ABS, температуры : стол 110, сопло 270. Перезагрузился на сток и запустил тот же файл, зависло с ошибкой E0011.

[image]
[image]

cmenard, 2024-06-05:
Not sure if it's related, but I had similar issues with longer prints.  Found that adding G92 E0 to the before layer change gcode has solved it.  The profiles in ORCA were missing that.  After adding it I haven't had any more errors or hangs while printing.

consp, 2024-06-07:
The timeout issues are due to memory exhaustion, my guess would be that klipper runs out of memory due to something specific in the gcode file since it's both on original and modded versions.

What's the size of the gcode file?

myxxx323, 2024-06-07:
> The timeout issues are due to memory exhaustion, my guess would be that klipper runs out of memory due to something specific in the gcode file since it's both on original and modded versions.
> 
> What's the size of the gcode file?

Various models, sizes from 10M to 50M

Разные модели, размер от 10М до 50М

thhdragon, 2024-06-09:
make sure you have arcs disabled in orca

myxxx323, 2024-06-09:
> make sure you have arcs disabled in orca

arcs disabled

ivask, 2024-06-15:
> Not sure if it's related, but I had similar issues with longer prints. Found that adding G92 E0 to the before layer change gcode has solved it. The profiles in ORCA were missing that. After adding it I haven't had any more errors or hangs while printing.

thanks, it works

jtenniswood, 2024-06-23:
I had issues after adding the webcam, I turned down the frame rate and it's been solid ever since. 

Probably not related but might help someone out.

consp, 2024-06-23:
> I turned down the frame rate and it's been solid ever since.

Since it looks like a memory issue I'm not surprized. Not running a gui will help as well. 

More swap should sound like it might help but due to how the swap works it might actually not matter if you would increase the swap file size as it cannot keep up swapping out memory with respect to the requested amount.

Not really have a good solution here as to try to save as much as possible. Since it happens on the stock firmware as well it might be worth investigating if it's a klipper issue or something else but I've not seen it happen yet.

cyberpods, 2024-06-25:
I found using Guppyscreen instead of Kilpperscreen helped my situation with this error, but it still happens once and a while.

consp, 2024-06-27:
> I found using Guppyscreen instead of Kilpperscreen helped my situation with this error, but it still happens once and a while.

Guppy uses about 24mb less memory than klipperscreen which causes this error to show up later. Using only the webinterfaces (no guid) will save you another 8-10mb (give or take a few mb's) which is a bit better than the flashforge OEM software.

Do you have a gcode file with which this happens (and are maybe willing to share), and do you know when it happens in the print (or before/after)?

If we can figure out where it starts hogging memory/cpu then we can figure out what the problem is and (maybe) fix it.

I usually print smaller parts and have not yet encoutered it other than during calibration which is a known issue.

harvinchan10, 2024-06-27:
Got this error as well.

I switched to no-GUI and so far so good. But haven’t really printed a big gcode yet since I switched.

not a solutiom but just wanna share my observations

the crash usually happens for me when I access Mainsail simultaneously on my computer + mobile phone. And it will surely crash if I open the webcam stream on another browser tab (accessing via http://192…:8080/?action=stream). Basically, if I stream the webcam on multiple devices it most likely exhaust the RAM as mentioned here.

i noticed on the OEM firmware, they only allow one video stream. Say if I’m streaming the webcam on my computer. I wont be able to stream on another computer if I access the Local ip stream (http://192…:8080/?action=stream). It does work on the flashforge app though since it goes through their server maybe.

consp, 2024-07-01:
> i noticed on the OEM firmware, they only allow one video stream. 

That might be a good idea to implement as well.

Arnaud3767, 2024-07-07:
This has been happening to me since the orcaslicer 2.1 update too.
New features may result in more gcode processing?

myxxx323, 2024-07-08:
I haven't used gyroid filling for a while now, and there are fewer errors.

Некоторое время не использую заполнение гироид, ошибок стало меньше.

jtenniswood, 2024-07-08:
I switched to Guppyscreen and that fixed the memory issues. I wish they had spend a little more on the hardware (or that it was upgradable!)

myxxx323, 2024-07-08:
> I switched to Guppyscreen and that fixed the memory issues. I wish they had spend a little more on the hardware (or that it was upgradable!)

There is also a problem on the drain. I monitored the memory on klipperscreen, at the time of the error it was not overloaded, it felt like the buffer was filling up.

Проблема есть и на стоке. Я наблюдал за памятью на klipperscreen, во время ошибки она не была перегружена, ощущение что заполняется буфер.

Arnaud3767, 2024-07-08:
Do you have to uninstall everything to install Guppyscreen?

jtenniswood, 2024-07-08:
I just flashed it over the top, and worked perfectly out of the box.
If you have made changes to the config files, you might want to grab a backup (as it will replace them).
Otherwise just the same as.

consp, 2024-07-09:
> Do you have to uninstall everything to install Guppyscreen?

No, but do check if you made changes to the configs if they are still there (hint: always backup).

consp, 2024-07-09:
> it felt like the buffer was filling up.

What happens is the memory starts swapping out to flash to avoid memory saturation. The lilely  problem (in my oppinion) is the flash cannot keep up and the kernel most likely will either hit an outstanding page limit or simply wait for pages to come available causing the memory access to have massive write delay penalties resulting in klipper timing out.

myxxx323, 2024-07-09:
Сегодня целый день печатаю, ни одного сбоя. Использую гироид, арки, все ок. Изменил в конфиге эти значения baud: 250000, было 230400 :

[mcu eboard]
serial: /dev/ttyS1
baud: 250000
restart_method: command

[mcu]
serial: /dev/ttyS5
baud: 250000
restart_method: command

Проверьте.

consp, 2024-07-18:
I don't understand Cyrillic so my guess is you think the baud rate mismatch is the problem. 8% is a bit much but not outside the realm of possibilities of most chips. On the other hand you cannot rule it out either way.

Is this the table for the NC clone chip or for the STM32?

myxxx323, 2024-07-18:
> I don't understand Cyrillic

Воспользуйтесь переводчиком, весь мир так делает :)

KaruroChori, 2024-07-19:
> Воспользуйтесь переводчиком, весь мир так делает :)

Fair, but following this logic, most messages here would rapidly devolve in a mess of German, Italian, French, Mandarin, Russian and more.  
You cannot expect each of us to need a multi-lingual translator just to understand what someone else wrote in this repo. 
It is perfectly fine if you want to attach the original message as you did at the beginning, so that there is a clear reference in case the machine translation is a bit off.
But please, provide an English translation as well, or you are just making the experience for everyone else worse.

myxxx323, 2024-07-19:
> Fair, but following this logic, most messages here would rapidly devolve in a mess of German, Italian, French, Mandarin, Russian and more. You cannot expect each of us to need a multi-lingual [...]

ok, I agree with you
ок, согласен с Вами

consp, 2024-07-22:
Languages aside:

> Is this the table for the NC clone chip in the AD5M or for the original STM32?

The reason I asked is that sometimes the PLLs on clone chips are a bit different, as are the sampling characteristics of the peripherals. Might be worth a try though. I'll give it a try to see what happens. 

I do suspect the baud rate increase is more likely to avoid the timeout issue than the baud rate error. Unless the uart interrupt implementation is wonky and causes weird sideffects. Sounds like a nice race condition somewhere. If it's a simple 4 wire uart connection or only pcb traces solution 0.5/1M might even work fine without issues.

AJolly, 2024-07-27:
G92 E0 does not solve the issue for me (including on guppyscreen)   
I do wish it was easier to just run klipper off an external box.

skylatian, 2024-07-29:
bump, this also happens to me when hitting the "print again" button on klipperscreen or in the UI after a print finishes. not a huge issue overall, but definitely not ideal

consp, 2024-08-01:
> I do wish it was easier to just run klipper off an external box.

That might solve it yes.

> bump, this also happens to me when hitting the "print again" button on klipperscreen or in the UI after a print finishes. not a huge issue overall, but definitely not ideal

Good to know, was this a relatively small print or a large one?

skylatian, 2024-08-01:
it just happened again, both with small prints (one was 20mins, the other 5). This time, it happened when I went to print another file that was already uploaded to the printer. It seems to only happen when you try to start another print immediately after printing something else, I'm pretty sure that it hasn't happened when I go to change filaments or home the machine after a print, but I could be wrong. Happy to provide more extensive logs, but this is what shows in the console:

```
$ HYPERLAPSE ACTION=STOP
03:55:38 PM 
$ _SET_TIMELAPSE_SETUP ENABLE=True VERBOSE=False PARK_ENABLE=False PARK_POS=back_left CUSTOM_POS_X=10.0 CUSTOM_POS_Y=10.0 CUSTOM_POS_DZ=0.0 TRAVEL_SPEED=100 RETRACT_SPEED=15 EXTRUDE_SPEED=15 RETRACT_DISTANCE=1.0 EXTRUDE_DISTANCE=1.0 PARK_TIME=0.1 FW_RETRACT=False
03:55:27 PM 
// Klipper state: Disconnect
03:55:27 PM 
$ FIRMWARE_RESTART
03:54:46 PM 
!! MCU 'mcu' shutdown: Timer too close
03:54:46 PM 
// MCU 'mcu' shutdown: Timer too close
// This often indicates the host computer is overloaded. Check
// for other processes consuming excessive CPU time, high swap
// usage, disk errors, overheating, unstable voltage, or
// similar system problems on the host computer.
// Once the underlying issue is corrected, use the
// "FIRMWARE_RESTART" command to reset the firmware, reload the
// config, and restart the host software.
// Printer is shutdown
03:54:46 PM 
!! Homing failed due to printer shutdown
03:54:46 PM 
!! Homing failed due to printer shutdown
03:54:46 PM 
!! Homing failed due to printer shutdown
03:54:45 PM 
// Klipper state: Shutdown
```

draskostar, 2024-11-03:
Is there anyone that can confirm this? I am on Guppy screen but Klipper screen is my favorite so I am thinking to give it a try.

https://www.reddit.com/r/FlashForge/comments/1cu1rnq/psa_disable_arc_fitting_in_orcaslicer_on_ad5m/

myxxx323, 2024-11-04:
Switched to stock firmware, removed the mod, after updating the firmware to 2.7.6 - not a single crash, printed about 5 kg of plastic. Printed for 20 minutes and for 6 hours.

Перешел на стоковую прошивку, удалил мод, после обновления прошивки на 2.7.6 - ни одного вылета, отпечатал порядка 5кг пластика. Печатал и на 20 минут и на 6 часов.

draskostar, 2024-11-10:
> Switched to stock firmware, removed the mod, after updating the firmware to 2.7.6 - not a single crash, printed about 5 kg of plastic. Printed for 20 minutes and for 6 hours.
>
> Перешел на стоковую [...]

I can confirm that. You have all my respect. Thank you very much for sharing the solution.

Berkano24, 2024-11-29:
Same on my side.. it even appeared after printing has been finished.

xblax, 2024-11-29:
If think the most important thing, in order even try get to the bottom of this issue, would be if someone has a G-code that does very reliably trigger the issue, without needing a long print.

My guess, why it happens with the mod but non on stock, is that it's somehow related to Moonraker putting additional load onto Klipper (Stock has no Moonraker) in combination with the shortage of memory.

If this issue was easily reproducible, one could at least try to understand the root cause and see how it can be mitigated.

Berkano24, 2024-12-02:
I see it now on a nearly daily base, even sometimes after some short prints (don't believe it has something to do with the memory). myxxx323 already mentioned that it might have something to do with the baud rate. Did somone follow this approach in the meantime?

1Dudka31, 2024-12-26:
> Теперь я вижу это почти каждый день, даже иногда после коротких распечаток (не думаю, что это как-то связано с памятью). myxxx323 уже упоминал, что это может быть как-то связано со скоростью передачи данных. Кто-нибудь уже пробовал этот способ?

я попробовал
у меня было 2 модели которые не могли распечататься
с 1 сработала
другая на сантиметр позже но все же остановилось
в общем на 250 у меня проблема решилать 50/50

omrta-dev, 2025-01-12:
It's probably just a RAM / memory leak issue. What fixed my consistent crashes was disabling timelapses / removing the camera. Not a single crash when not running a camera / timelapses

myxxx323, 2025-01-13:
В новой версии прошивки 3.1.3 добавили nim - библиотеку, которая позволяет ускорить код на питоне. Поменялось
-  library 2.1.0-2.3.0
- software 2.7.9->3.1.3 , пока проблем на стоке не обнаружилось, возможно ли обновить мод с этими библиотеками?

draskostar, 2025-02-01:
Can this helps anyhow?
https://www.reddit.com/r/FlashForge/s/i3R7G2nLr2

AR1972, 2025-03-01:
has anybody tried to swap out the MCU with the Allwinner T113-S4 256MB RAM variant? Would there need to be bootloader or software modifications to detect the increased RAM?