xblax/flashforge_ad5m_klipper_mod #207: guppyscreen input shaper not saving

Buko84, 2024-06-13:
Hi, just upgraded to 00.05 beta with guppyscreen it's not saving input shaper, it's going round and round forever and there is such info in console as on photo
[image]

consp, 2024-06-14:
Looks like the guppyscreen cfg in not included by klipper. I'll have a look.

SAVE_INPUT_SHAPER is not a command, looks like a bug in guppyscreen itself.

consp, 2024-06-14:
I've created a pull request for the save/set change. https://github.com/ballaswag/guppyscreen/pull/105

xblax, 2025-09-16:
This *should* be fixed since the mod was rebuild against the latest branch for the mod, however did not test this.

I'm closing the GuppyScreen issues, since it's unlikely they will be worked at by someone and upstream development has also stalled. I added some information about Guppy Screen "broken" state to [Known Issues](https://github.com/xblax/flashforge_ad5m_klipper_mod/blob/master/docs/LINUX.md), so that it's clear it can be used as-is but will not be worked on at the moment.

closed 2025-09-16 by xblax