xblax/flashforge_ad5m_klipper_mod #213: Module libcamera unavailable, import failed

dainbrump, 2024-07-01:
Using the Adventurer5M image (not the pro since I don't have the pro) I can't get the official FF camera to load in mainsail. Kinda seems to work in fluidd. Checked the moonraker log and found:

```
Module libcamera unavailble, import failed
Klipper Version: v0.11.0-4-g4d24de080-AD5M-20240506
2024-07-01 15:09:12,584 [authorization.py:_prune_conn_handler()] - Trusted Connection Expired, IP: 127.0.0.1
2024-07-01 15:09:12,585 [authorization.py:_prune_conn_handler()] - Trusted Connection Expired, IP: 192.168.86.170
2024-07-01 15:42:02,400 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.86.170
2024-07-01 16:11:41,615 [application.py:prepare()] - Upload Request Received from 192.168.86.170
Content-Type: 
2024-07-01 16:11:42,386 [application.py:prepare()] - Upload Request Received from 192.168.86.170
Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryosFjpci7MJJ9eqt8
2024-07-01 16:11:42,565 [application.py:post()] - Processing Uploaded File: moonraker.conf
2024-07-01 16:11:42,595 [application.py:log_request()] - 201 POST /server/files/upload (192.168.86.170) [_TRUSTED_USER_] 211.12ms
2024-07-01 16:11:43,633 [websockets.py:on_close()] - Websocket Closed: ID: 2813024464 Close Code: 1000, Close Reason: None, Pong Time Elapsed: 4.86
2024-07-01 16:11:43,747 [websockets.py:write_to_socket()] - Websocket closed while writing: 2812569488
2024-07-01 16:11:44,423 [websockets.py:on_close()] - Websocket Closed: ID: 2812569488 Close Code: 1001, Close Reason: Server Shutdown, Pong Time Elapsed: 5.31
2024-07-01 16:11:44,448 [klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed
2024-07-01 16:11:44,493 [database.py:close()] - Database statistics:
```

I'm assuming I did a dumb but can't figure out what dumb I did.

xblax, 2024-07-01:
Hi @dainbrump , libcamera support for Moonraker has different purpose, we don't need it in our case. Mainsail and Fluidd directly connect to ustreamer via HTTP.

Try the following settings in Mainsail (with your IP):

[image]

I assume you had tried the "MJPEG-Streamer" mode. That mode does not work with Mainsail and ustreamer. It's a bug/issue in Mainsail.

dainbrump, 2024-07-01:
Doh! I knew I did a dumb. Thanks. That got me picture. I believe I read somewhere else there was framerate wonkiness. Currently getting frames by postal service it seems. I'll take another look at those other github issues I stumbled across trying to figure this one out.

closed 2024-07-01 by dainbrump