xblax/flashforge_ad5m_klipper_mod #217: Configuring ERCF But need to update Klipper. Thoughts?

f00d4tehg0dz, 2024-07-07:
Hi There, 

I have built an ERCF and configured the buildroot SDK and subsequent variants to support Happy Hare. After configuring Happy Hare, I went to restart the Klipper firmware and received a few Toolhead sensor attributes missing inside the Klipper installation.  Happy Hare appended some configurations into those, which leads me to believe that the older version of Klipper is causing this issue. 

Any recommendations or steps to update Klipper to the latest built while still maintaining @xblax  configurations?

Thanks!

consp, 2024-07-08:
Does it need to update the kippy (python) part or the MCU firmware?

First: Doable if it can be done with v0.11 firmware. You need to replace the python files you need and possibly create gluelogic to be compatible.
Second: Not possible due to missing configuration files in the official flashforge klipper repository (you cannot get a working build of klipper firmware with it).

f00d4tehg0dz, 2024-07-08:
> Does it need to update the kippy (python) part or the klipper firmware?
> 
> First: Doable if it can be done with v0.11 firmware. You need to replace the python files you need and possibly create gluelogic to be compatible. Second: Not possible due to missing configuration files in the official flashforge klipper repository (you cannot get a working build of klipper firmware with it). This PITA was obvious from the start and the only way to do it (at this time as far as I know) is by replacing the STM clone chip with a genuine one.

Ah, I see, and yes, Kippy needs to be updated. So should be doable to your First comment. 
I believe the firmware is fine as is. I'll report back

Thank you!

f00d4tehg0dz, 2024-07-12:
Good to go ( i believe).
I had to use the latest klipper version available from this mod, and then leverage Happy Hare git commit [7915199748f38b8564ec970c5aa7b510cdffa914](https://github.com/moggieuk/Happy-Hare/commit/7915199748f38b8564ec970c5aa7b510cdffa914) . This commit is before major toolhead changes happened from Klipper.

closed 2024-07-12 by f00d4tehg0dz

consp, 2024-07-18:
Marked as completed but for reliability sake, can you report back after printing a while? I expect no problems but it's always nice to know you actually had none!