xblax/flashforge_ad5m_klipper_mod #219: heating halts before reaching set temperature

skylatian, 2024-07-09:
When setting either bed or extruder temps via Fluidd or Klipperscreen, the heater stops after ~11 seconds (repeatably). The temperature requested does not matter (I could set it to hold the current temperature, and the heaters will still turn off). 

This behavior occurs even after a full power-off reboot or a firmware/klipper restart. This did not appear to be happening on the previous version. It also still prints files correctly, this just makes it a nightmare to load/unload filament.

Here's a snippet of klipper.log from the line where heating begins to where it stops - I can't find anything unusual.
<details>
  <summary>klipper.log</summary>

```
Stats 37183.0: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000022 mcu_task_stddev=0.000013 bytes_write=27113052 bytes_read=9718875 bytes_retransmit=9 bytes_invalid=0 send_seq=500688 receive_seq=500688 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999799 eboard: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000008 bytes_write=11459208 bytes_read=3744849 bytes_retransmit=9 bytes_invalid=0 send_seq=230424 receive_seq=230424 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999847 adj=144000017 heater_bed: target=0 temp=26.2 pwm=0.000 Load_Cell: temp=817.0 TVOC_Level: temp=26.0  sysload=0.12 cputime=1943.685 memavail=27396 print_time=37187.853 buffer_time=0.135 print_stall=0 extruder: target=200 temp=28.9 pwm=0.000
Heater extruder approaching new target of 200.000
Stats 37184.0: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000022 mcu_task_stddev=0.000013 bytes_write=27113088 bytes_read=9719078 bytes_retransmit=9 bytes_invalid=0 send_seq=500691 receive_seq=500691 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999811 eboard: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000009 bytes_write=11459245 bytes_read=3744933 bytes_retransmit=9 bytes_invalid=0 send_seq=230427 receive_seq=230427 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999848 adj=143999971 heater_bed: target=0 temp=26.3 pwm=0.000 Load_Cell: temp=798.0 TVOC_Level: temp=26.0  sysload=0.12 cputime=1943.727 memavail=27396 print_time=37187.853 buffer_time=0.000 print_stall=0 extruder: target=200 temp=28.9 pwm=1.000
Stats 37185.0: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000022 mcu_task_stddev=0.000013 bytes_write=27113094 bytes_read=9719253 bytes_retransmit=9 bytes_invalid=0 send_seq=500692 receive_seq=500692 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999810 eboard: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000009 bytes_write=11459251 bytes_read=3745009 bytes_retransmit=9 bytes_invalid=0 send_seq=230428 receive_seq=230428 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999847 adj=143999821 heater_bed: target=0 temp=26.3 pwm=0.000 Load_Cell: temp=798.0 TVOC_Level: temp=26.0  sysload=0.11 cputime=1943.761 memavail=27420 print_time=37187.853 buffer_time=0.000 print_stall=0 extruder: target=200 temp=31.0 pwm=1.000
Stats 37186.0: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000022 mcu_task_stddev=0.000013 bytes_write=27113100 bytes_read=9719413 bytes_retransmit=9 bytes_invalid=0 send_seq=500693 receive_seq=500693 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999810 eboard: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000009 bytes_write=11459257 bytes_read=3745070 bytes_retransmit=9 bytes_invalid=0 send_seq=230429 receive_seq=230429 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999845 adj=143999874 heater_bed: target=0 temp=26.3 pwm=0.000 Load_Cell: temp=798.0 TVOC_Level: temp=26.0  sysload=0.11 cputime=1943.801 memavail=27316 print_time=37187.853 buffer_time=0.000 print_stall=0 extruder: target=200 temp=35.3 pwm=1.000
Stats 37187.0: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000022 mcu_task_stddev=0.000013 bytes_write=27113112 bytes_read=9719606 bytes_retransmit=9 bytes_invalid=0 send_seq=500695 receive_seq=500694 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999810 eboard: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000009 bytes_write=11459269 bytes_read=3745131 bytes_retransmit=9 bytes_invalid=0 send_seq=230431 receive_seq=230430 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999844 adj=143999876 heater_bed: target=0 temp=26.3 pwm=0.000 Load_Cell: temp=798.0 TVOC_Level: temp=26.0  sysload=0.11 cputime=1943.838 memavail=27340 print_time=37187.853 buffer_time=0.000 print_stall=0 extruder: target=200 temp=40.3 pwm=1.000
Stats 37188.0: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000022 mcu_task_stddev=0.000013 bytes_write=27113118 bytes_read=9719809 bytes_retransmit=9 bytes_invalid=0 send_seq=500696 receive_seq=500696 retransmit_seq=2 srtt=0.002 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999809 eboard: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000009 bytes_write=11459291 bytes_read=3745228 bytes_retransmit=9 bytes_invalid=0 send_seq=230433 receive_seq=230433 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999839 adj=143999869 heater_bed: target=0 temp=26.3 pwm=0.000 Load_Cell: temp=798.0 TVOC_Level: temp=26.0  sysload=0.11 cputime=1943.877 memavail=27380 print_time=37187.853 buffer_time=0.000 print_stall=0 extruder: target=200 temp=47.3 pwm=1.000
Stats 37189.0: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000022 mcu_task_stddev=0.000013 bytes_write=27113124 bytes_read=9719959 bytes_retransmit=9 bytes_invalid=0 send_seq=500697 receive_seq=500697 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999808 eboard: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000009 bytes_write=11459297 bytes_read=3745302 bytes_retransmit=9 bytes_invalid=0 send_seq=230434 receive_seq=230434 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999838 adj=143999890 heater_bed: target=0 temp=26.3 pwm=0.000 Load_Cell: temp=798.0 TVOC_Level: temp=26.0  sysload=0.11 cputime=1943.913 memavail=27328 print_time=37187.853 buffer_time=0.000 print_stall=0 extruder: target=200 temp=52.7 pwm=1.000
Stats 37190.0: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000022 mcu_task_stddev=0.000013 bytes_write=27113130 bytes_read=9720146 bytes_retransmit=9 bytes_invalid=0 send_seq=500698 receive_seq=500698 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999806 eboard: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000009 bytes_write=11459303 bytes_read=3745363 bytes_retransmit=9 bytes_invalid=0 send_seq=230435 receive_seq=230435 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999836 adj=143999918 heater_bed: target=0 temp=26.3 pwm=0.000 Load_Cell: temp=798.0 TVOC_Level: temp=26.0  sysload=0.11 cputime=1943.947 memavail=27400 print_time=37187.853 buffer_time=0.000 print_stall=0 extruder: target=200 temp=58.1 pwm=1.000
Stats 37191.0: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000022 mcu_task_stddev=0.000013 bytes_write=27113136 bytes_read=9720321 bytes_retransmit=9 bytes_invalid=0 send_seq=500699 receive_seq=500699 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999805 eboard: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000009 bytes_write=11459309 bytes_read=3745439 bytes_retransmit=9 bytes_invalid=0 send_seq=230436 receive_seq=230436 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999838 adj=143999934 heater_bed: target=0 temp=26.3 pwm=0.000 Load_Cell: temp=817.0 TVOC_Level: temp=26.0  sysload=0.11 cputime=1943.982 memavail=27412 print_time=37187.853 buffer_time=0.000 print_stall=0 extruder: target=200 temp=65.1 pwm=1.000
Stats 37192.0: gcodein=0  mcu: mcu_awake=0.002 mcu_task_avg=0.000022 mcu_task_stddev=0.000013 bytes_write=27113142 bytes_read=9720481 bytes_retransmit=9 bytes_invalid=0 send_seq=500700 receive_seq=500700 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999805 eboard: mcu_awake=0.001 mcu_task_avg=0.000012 mcu_task_stddev=0.000009 bytes_write=11459331 bytes_read=3745505 bytes_retransmit=9 bytes_invalid=0  [... cut ...] task_avg=0.000012 mcu_task_stddev=0.000009 bytes_write=11459356 bytes_read=3745654 bytes_retransmit=9 bytes_invalid=0 send_seq=230441 receive_seq=230441 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999834 adj=144000009 heater_bed: target=0 temp=26.2 pwm=0.000 Load_Cell: temp=798.0 TVOC_Level: temp=26.0  sysload=0.11 cputime=1944.105 memavail=27420 print_time=37198.807 buffer_time=0.079 print_stall=0 extruder: target=0 temp=82.2 pwm=0.000
```
</details>

xblax, 2024-07-10:
Is there something printed in Klipper console? Never had this happen for me.

skylatian, 2024-07-14:
There wasn't, it was very odd! The behavior went away after updating to v00.05, must've been related to one of the bugfixes made with that release.

closed 2024-07-14 by skylatian