xblax/flashforge_ad5m_klipper_mod #221: issues / ideas for improving the filament runout (macro?) process

DougJoseph, 2024-07-10:
Thanks again for such a great mod!

Whenever the filament runout situation is triggered, sometimes the user is not able to address the situation quickly. 

Two things then happen that are problematic.

The nozzle temp goes to naught, and there is no way in the modal dialog box to heat it back up (and no way to access temp controls while in this modal mode), that I could find, even using both Mainsail and Klipperscreen.

The stepper motors disengage, and it's way too easy to move the parked print head while attempting a filament load. It is possible to escape out of the modal dialog box completely, for purposes of re-homing and then resuming, but for many users their fear may well be as mine, that (1) homing may somehow interrupt or erase or derail the resume step, and (2) no idea how to manually invoke the resume process even if it will work.

This all happened for me just now, and it was "scary" / perturbing, and for a while I was flummoxed, yet I was eventually thrilled to learn that if I escape out of the filament runout modal dialog, and manually raise the nozzle temp, manually load filament (which resulted in accidentally moving the print head out of its parked location, manually home (because of said accidental movement), and manually type "resume" in the console, which all seemed to work wonderfully.

In light of all this, it seems the dialog box could benefit from some added options / added conveyed details.

One other note: the default parking location is right up against the riser wall of my DIY enclosure, in a place that's practically impossible to get to for a reload. Can I revise that location to, perhaps, the center of the bed? If so, where would I look to do that?

Thanks again for being awesome!

xblax, 2024-07-10:
Is your report based on this recent discussion here or is this just a coincidence? 
https://github.com/xblax/flashforge_ad5m_klipper_mod/discussions/220

As I said there this happens after the 10min idle timeout. Maybe we can try to close the dialog when the idle timeout kicks in and show something different.

The park position was chosen to avoid oozing filament onto the print while waiting for the change and loading new filament. It can be changed by adjusting the values in macros.cfg:
```
variable_custom_park_x    : -105.0  ; custom x position; value must be within your defined min and max of X
variable_custom_park_y    : -105.0  ; custom y position; value must be within your defined min and max of Y
variable_custom_park_dz   : 10.0    ; custom dz value; the value in mm to lift the nozzle when move to park position
```

DougJoseph, 2024-07-10:
> Is your report based on this recent discussion here or is this just a coincidence?

I had not even seen that other discussion! So sorry for doubling up on things in separate threads.

> As I said there this happens after the 10min idle timeout. Maybe we can try to close the dialog when the idle timeout kicks in and show something different.

Cooling off the bed and the nozzle after a few minutes is wise. 

What would be helpful is a way within the modal dialog box to get them heated back up again. Maybe include logic for that in the load and unload buttons. For that matter, resuming would also need things to be heated back up. Also, would be helpful to present some type of feedback that lets the user know that things cooled off, but we’re heating them back up again or some such.

And maybe add a homing button as well, or at least build re-homing into the resume button, and perhaps mention in the modal dialog box, that resuming will do a homing process first.

> The park position was chosen to avoid oozing filament onto the print while waiting for the change and loading new filament. It can be changed by adjusting the values in macros.cfg:

Good thinking about the oozing, especially related to reloading. I guess for me parking it at the back (right) wood be better than at the front (left). Middle is not good suggestion. :-)

DougJoseph, 2024-07-10:
Maybe after the cooldown happens, change the options in the modal dialog box to say, for example, “reheat and load”, “reheat and unload,” “reheat and resume,” etc.

DougJoseph, 2024-07-10:
Thanks for the helpful info on the park location! I will look into getting that edited.

xblax, 2024-07-10:
> What would be helpful is a way within the modal dialog box to get them heated back up again. Maybe include logic for that in the load and unload buttons. For that matter, resuming would also need [...]

I believe there is not really a way to know what temperatures the print was using in the macros. What I could imagine is to close that dialog and show a simple dialog that the printer is idle and must be resumed manually.

DougJoseph, 2024-07-10:
> I believe there is not really a way to know what temperatures the print was using in the macros. What I could imagine is to close that dialog and show a simple dialog that the printer is idle and must be resumed manually

Yes, this would be helpful, especially if it mentioned (for benefit of new users) that short list of things that need done (heat up, reload, home, resume), and for the benefit of users not real familiar with Klipper, that typing "Resume" in the console will do the trick.

Speaking of which, I should try to figure out if there is a way to manually resume over in Klipper screen. Is there a button for that, or is there a Klipper screen version of the console?