xblax/flashforge_ad5m_klipper_mod #224: z_offset get ignored when saved in overrides, gets respected with original

DougJoseph, 2024-07-15:


DougJoseph, 2024-07-16:
It may be that I don't properly understand what the Z_Offset value is used for. 

According to one Redditor in this post:  

https://www.reddit.com/r/klippers/comments/h0cb1e/comment/ftlnhxk/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

He suspects the value is only used during probing, and since I don't always re-probe on every print job (I home to the endstop, not a touch off with the nozzle probing the bed) it may well be that what the machine perceives based on the end of the "Z calibrate" process (in which I manually lower the nozzle until the paper is barely scratched by the nozzle) is the only thing it goes by, until the next time it probes to the bed. But how would I invoke a probe to the bed that is not yet another Z calibrate with the paper scratch process again?

There's possibility for a lot of variance between each manual paper scrape test. The first time I did it on this new (third) AD5M, I did not pay attention to get it at the "barely scrape" point because I thought, "I'll just edit in the config file." I had it pretty snug, and it saved an auto-calculated value of -0.16. After reading the above Reddit post about suspecting it only gets referenced during probing, I repeated Z calibrate, and I paid attention, getting it to just barely scrape the paper, and it saved a new value of -.31. However, on my next attempt at a print job, that still made absolutely no difference. The nozzle was too high.

At this point, experienced Klipper users reading this are thinking, "Why don't you just use fine tuning?"  (aka baby stepping).

I'll not only explain why I had not been using fine tuning / baby stepping, and what I finally just now learned, after months of using Klipperscreen on the AD5M.

REASON: The Klipperscreen GUI on the AD5M does not show any "Save values" button on the fine tuning (baby stepping) screen. When you go back out of the fine tuning screen, there is **still** "Save values" button — no option for updating things so you don't have to keep doing it.

WHAT I JUST SAW for the first time: after I cancelled the job, I finally got shown a button to save values (presumably from the fine tuning). I did not at that time save the values because I had been a bit over zealous, and I wanted to redo it. I am in hopes that the same button gets shown after the job completes, at which time I plan to punch the button with prejudice and careful attention to seeing it done.

This leads me to: if the "Values saved" from fine tuning don't get stored in an adjustment to the Z_Offset value, then where in the Klipper world do those fine tuning values get saved? Because that's what I should have been editing!

Thanks for any clarity, corrections, or advice anyone can give.

PS: If I eventually prove and conclude that fine-tuning + saving values actually accomplished what I'm after, I will mark this post as the solution! :-)

DougJoseph, 2024-07-16:
**I may be onto something.  In macros.cfg, the macro section labelled**

`[gcode_macro START_PRINT]`

... contains this line of code:

`SET_GCODE_OFFSET Z=0.0  ; reset the G-Code Z offset (adjust Z offset if needed)`

And according to these two posts:
* https://klipper.discourse.group/t/klipper-not-applying-z-offset-to-future-prints/13317/18
* https://klipper.discourse.group/t/klipper-not-applying-z-offset-to-future-prints/13317/19

**... the above line of code in the macro, invalidates my saved z_offset in printer.cfg.**

xblax, 2024-07-16:
Hey @DougJoseph, I think you are mixing up some things:

Probe Z-Offset and G-Code Z-Offset are separate offsets. G-Code Z-Offset is used to fine-adjust for a specific print, i.e. to adjust offset during print. Resetting the G-Code Z-Offset before printing has nothing to with Probe Z-Offset.

Probe Z-Offset is taken into Account when Probing. After you change this value, then re-do your bed mesh - otherwise it does not have any effect.

 Z_OFFSET_APPLY_PROBE should not go into START_PRINT macro, this would just mess things up.

DougJoseph, 2024-07-16:
@xblax Thanks! I will change my approach and see if I can get success.

DougJoseph, 2024-07-16:
> Probe Z-Offset is taken into Account when Probing. After you change this value, then re-do your bed mesh - otherwise it does not have any effect.

Dude, you are awesome. That did the trick. All I had to do was repeat the mesh probe process after setting the Z_Offset, and it finally has the nozzle at the height it needs to be! Thank you so much. 

If there is a way to mark your answer as the solution, I would like to.  I don't see how to do that yet.

closed 2024-07-16 by DougJoseph