xblax/flashforge_ad5m_klipper_mod #233: TMC extruder reports error: GSTAT 00000001

Kuckmalhierher-78, 2024-07-30:
Hello,
First of all, thank you for the Klipper variant for the AD5M.
For 2 days now I have been getting this error when printing. The AD5M always breaks off at different points during printing.
The cables look good and I have already changed the noozle.
I have also restored the factory settings without success.
Does anyone here have a solution?

My English is not good :-)

Greetings Sven

[image]

consp, 2024-08-01:
Never seen this one before.

Looking at the threads on the klipper fora it's a motor problem. Try switching to the OEM firmware and see if it happens there. I'm pretty sure it will but that will confirm it's most likely a EM problem and not software related.

Kuckmalhierher-78, 2024-08-01:
I tried the whole thing on a second Flashforge AD5M. Here too, the same error after a while.
The funny thing is that this happens with the filament TPU from Overture on both devices. With PLA everything works for several hours without errors.

So I think it's the filament and not the AD5M.

Best regards

SelfhostedPro, 2024-09-14:
Have you found any work around for working with the tpu? I’m getting the same error with the same filament

Kuckmalhierher-78, 2024-09-15:
I used a different filament manufacturer.
At the moment I get very good results with the TPU from YXPOLYER without error messages.

buffy1989, 2024-10-15:
Hi, 
I have the same Problem with Klipper. Then I have uninstall it and on the official i haven't the Problem.
best regards