xblax/flashforge_ad5m_klipper_mod #235: USB ports not working?

dangerousderek, 2024-07-30:
Sorry to put this here, but not sure who else to ask. I soldered some headers onto the spots marked as USB on the stock AD5M motherboard, and no matter what I try to hook up to them they don't seem to work. I am pretty sure I soldered them correctly and the camera I was trying to plug in works when plugged into the display camera port, but not the motherboard? Any ideas?
Thanks!

consp, 2024-08-01:
Haven't tried it yet but some resistors might be missing (or 0 ohm bridges). Also: Not all USB ports are connected to the MCU.

Which one did you connect it to? And do you see anything at all in the syslog?

xblax, 2025-09-16:
According to the information I have, only USB0 on the mainboard can be used for additional devices. Other unsoldered headers on the mainboard are not connected or already used for Wifi module.

closed 2025-09-16 by xblax