xblax/flashforge_ad5m_klipper_mod #248: Klipper parser (perhaps) error

Shadow-FW, 2024-09-17:
Hello. The printer stopped printing with the note "Canceled". After reviewing the console output, I found that it gave the following information: Unable to parse move 'G1 **X-2-4.909** Y-23.009 I-.678 J-.51 E.03589'
Unknown command:"GG2"
After scrolling through the GCODE file in the printer itself, I found places where these errors are: "G1 **X-2-4.909** Y-23.009 I-.678 J-.51 E.03589
**GG2** X-19.629 Y-10.963 I4.051 J-2.784 E.03089"
There are no such errors in the original file. I uploaded the same original file again, and there were no such errors. Apparently there are some errors in the Klipper parser, it would be good to update the Klipper version. Although there may be a mistake in something else, perhaps knowledgeable people will said some.

KaruroChori, 2024-09-17:
In the past we had syntax issues which were related to specific versions of Orcaslicer. It was not a klipper problem. 
Aside from that, sadly there is no easy way to update to a more modern version of klipper because of the microcontrolled used for the toolhead.

Shadow-FW, 2024-09-17:
Hi. Thanks for your reply. But the thing is that the original file from OrcaSlicer doesn't have these errors. They appeared only after I uploaded it into the printer.

KaruroChori, 2024-09-18:
Have you kept a copy of both files? How did you move it over?

Shadow-FW, 2024-09-18:
Yep, i save both files. Upload it?

KaruroChori, 2024-09-20:
Yes, please

Shadow-FW, 2024-09-20:
[Bug.zip](https://github.com/user-attachments/files/17072810/Bug.zip)
[Orig.zip](https://github.com/user-attachments/files/17072839/Orig.zip)
Orig from Orca, Bug from printer uploaded

vjcg92, 2025-03-15:
I'm having the same issue. I can print from a USB drive just fine, but it seems like the gcode is getting corrupted when sent and written to the internal flash.

david-we, 2025-10-06:
got the same issue stopped after hours in the print. corrupt gcode command.  

`Unable to parse move 'G1 X42.199 Y49.831 Z18.-3.631 Y-6.168 E.00697'`

newest klippermod version and orcaslicer 2.3.1-beta

is there a way to do a md5 checksum or something similar to recognize if the file is corrupt?

[image]

[image]