xblax/flashforge_ad5m_klipper_mod #249: Z offset not working

WatWall, 2024-09-28:
Today I've installed klipper mod again. Didn't change any setting, Set bed temp to 60, extruder to 120 and did a bed mesh calibration through guppyscreen and saved the profile at the end. 
Then I did a test print but z offset was way too high.
Also force leveling before the print does not make any difference.

WatWall, 2024-09-28:
I calculate Z offset as 0.25 (not -0.25) and save it but no change at all in z offset while printing

santahx, 2024-10-04:
Hey hier von PRINT_START
welches immer am anfang deines Gcodes ausgeführt wird

Muast bei macros unter print start nur deinen wert anpassen.
Mit Freundlichen Grüßen Nicolas Bissinger 

    SET_GCODE_OFFSET Z=0.0  ; reset the G-Code Z offset (adjust Z offset if needed)