xblax/flashforge_ad5m_klipper_mod #250: I have repeating crashes on my M5PRO, clipper crashes in the middle of the print

ShayGus, 2024-10-11:
Clipper crashes in the middle of the print.
Providinbg log.

Dumping stepper 'stepper_z' (mcu) 5 queue_step:
queue_step 0: t=111759778813 p=-87277 i=180000 c=-40 a=0
queue_step 1: t=111766978813 p=-87317 i=180000 c=-40 a=0
queue_step 2: t=111774178813 p=-87357 i=180000 c=-40 a=0
queue_step 3: t=111781378813 p=-87397 i=180000 c=-40 a=0
queue_step 4: t=111788578813 p=-87437 i=180000 c=-40 a=0
Dumping trapq 'toolhead' 1 moves:
move 0: pt=774.933450 mt=5.996000 sv=2.000000 a=0.000000 sp=(-105.000000,-105.000000,1.996000) ar=(0.000000,0.000000,-1.000000)
Requested toolhead position at shutdown time 776.218225: (-105.0, -105.0, -0.5735488236881374)
Stats 771.7: gcodein=0  mcu: mcu_awake=0.004 mcu_task_avg=0.000023 mcu_task_stddev=0.000012 bytes_write=38935 bytes_read=164664 bytes_retransmit=9 bytes_invalid=0 send_seq=3002 receive_seq=3002 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=144000328 eboard: mcu_awake=0.001 mcu_task_avg=0.000014 mcu_task_stddev=0.000009 bytes_write=28651 bytes_read=80926 bytes_retransmit=9 bytes_invalid=0 send_seq=2340 receive_seq=2340 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=143999724 adj=143999202 heater_bed: target=60 temp=68.0 pwm=0.000 Load_Cell: temp=152.0 TVOC_Level: temp=23.0 sd_pos=1876 sysload=0.48 cputime=49.801 memavail=28432 print_time=776.612 buffer_time=0.107 print_stall=0 extruder: target=0 temp=114.5 pwm=0.000
Probing failed due to printer shutdown
Probing failed due to printer shutdown
Probing failed due to printer shutdown
Probing failed due to printer shutdown
MCU 'mcu' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Script running error
Traceback (most recent call last):
  File "/root/printer_software/klipper/klippy/extras/gcode_button.py", line 41, in button_callback
    self.gcode.run_script(template.render())
  File "/root/printer_software/klipper/klippy/gcode.py", line 216, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/root/printer_software/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/root/printer_software/klipper/klippy/gcode.py", line 272, in cmd_default
    raise gcmd.error(self.printer.get_state_message()[0])
gcode.CommandError: MCU 'mcu' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

MCU 'mcu' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

virtual_sdcard on_error
Traceback (most recent call last):
  File "/root/printer_software/klipper/klippy/mcu.py", line 511, in _do_send
    return xh.get_response(cmds, self._cmd_queue, minclock, reqclock)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/klipper/klippy/serialhdl.py", line 324, in get_response
    raise error("Unable to obtain '%s' response" % (self.name,))
serialhdl.error: Unable to obtain 'trsync_state' response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/printer_software/klipper/klippy/extras/homing.py", line 246, in probing_move
[... 86 lines cut ...]
          ^^^^^^^^^^^^^^^^^^
  File "/root/printer_software/klipper/klippy/extras/probe.py", line 129, in _probe
    raise self.printer.command_error(reason)
gcode.CommandError: Probing failed due to printer shutdown

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/printer_software/klipper/klippy/extras/virtual_sdcard.py", line 268, in work_handler
    self.gcode.run_script(self.on_error_gcode.render())
  File "/root/printer_software/klipper/klippy/gcode.py", line 216, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/root/printer_software/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/root/printer_software/klipper/klippy/gcode.py", line 272, in cmd_default
    raise gcmd.error(self.printer.get_state_message()[0])
gcode.CommandError: MCU 'mcu' shutdown: Timer too close
This often indicates the host computer is overloaded. Check
for other processes consuming excessive CPU time, high swap
usage, disk errors, overheating, unstable voltage, or
similar system problems on the host computer.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

MakerViking, 2024-10-12:
I've had it happen too, after it being on for a week or so. Rebooted and hasn't happened again. I'm using the Guppy version, since the Klipper Screen version crashed during input shaping for me.

Berkano24, 2024-11-26:
have seen similar issues during my prints.

KaruroChori, 2024-12-31:
Duplicate of https://github.com/xblax/flashforge_ad5m_klipper_mod/issues/203

closed 2024-12-31 by KaruroChori