xblax/flashforge_ad5m_klipper_mod #272: Unable to parse move hours into print

clutchkick512, 2024-12-16:
I was a few hours into a print and it was canceled. I’m not at home monitoring remotely and have this error on my console.

Unable to parse move 'G1 X-7..115 Y-19.115 E.00267'
// Unknown command:"G1 .62"

fresh ad5 I just got it today, running the guppy screen version. Orcaslicer ad5 default settings.
[klipper.log](https://github.com/user-attachments/files/18146869/klipper.log)

quazono, 2024-12-28:
Greetings, I have also encountered this issue with ad5m guppy screen version 00.05. I believe this issue can manifest a few different ways.

1. Move out of range. The printer can attempt to move to some large number that is out of range. This happened to me initially, and I made a [reddit post](https://www.reddit.com/r/3Dprinting/comments/1hdsd0t/random_move_out_of_range/) detailing the issue.
 
2. Unparseable gcode as shown in original comment. It happened to me where the x value had 2 decimals, like: 23.765.4 (don't remember the exact number and did not save that file.

3.  I believe one time I noticed that mid layer it would switch parts, do a little bit, and then go back to the first part, extruding during the travel between parts. While this does cause print problems it will not stop the print as it is in bounds and parseable. 

I began to look into this problem as it kept happening. Originally I wondered if it was a slicer issue but I have since ruled that out a couple ways. Currently my hypothesis is it happens during file transfer. The latest time this has happened I sliced the file in orca, exported gcode to my computer, uploaded the gcode to my printer (but did not print, only upload), downloaded the gcode to my computer, and diffed the 2 versions (broken gcode is on the left). 
[image]

I would be happy to provide any more needed information :)

KaruroChori, 2024-12-31:
Yes, it is the same problem. I am closing this one as a duplicate. Sadly I was never able to replicate the issue myself. @clutchkick512, how are you transferring the file? Which client are you using?

closed 2024-12-31 by KaruroChori